flipperdevices/flipperzero-firmware #3794: NFC: Desfire Poller furi_check failed

s1rleblanc, 2024-07-18:
### Describe the bug.

Im trying to clone a NFC Hotel Card and every time i try to do that all the flipper crash

[image]

### Reproduction

1. Go to NFC
2. Read
3. Put card in flipper back
4. Crash

### Logs

```Text
25451 [D][NfcScanner] Found 5 children
25506 [D][Nfc] FWT Timeout
25508 [D][Nfc] FWT Timeout
25542 [D][Nfc] FWT Timeout
25574 [D][Iso14443_4aPoller] Read ATS success
25610 [D][Iso14443_4aPoller] Read ATS success
25641 [I][NfcScanner] Detected 1 protocols
25758 [W][ViewPort] ViewPort lockup: see applications/services/gui/view_port.c:245
25812 [I][Elf] Total size of loaded sections: 888
25815 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
25857 [I][Elf] Total size of loaded sections: 420
25859 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
25905 [I][Elf] Total size of loaded sections: 1804
25908 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
25953 [I][Elf] Total size of loaded sections: 2944
25956 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26000 [I][Elf] Total size of loaded sections: 284
26003 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26053 [I][Elf] Total size of loaded sections: 1356
26056 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26104 [I][Elf] Total size of loaded sections: 820
26107 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26150 [I][Elf] Total size of loaded sections: 552
26153 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26201 [I][Elf] Total size of loaded sections: 1396
26204 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26268 [I][Elf] Total size of loaded sections: 1608
26271 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26316 [I][Elf] Total size of loaded sections: 908
26319 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26372 [I][Elf] Total size of loaded sections: 1212
26375 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26426 [I][Elf] Total size of loaded sections: 1324
26429 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26478 [I][Elf] Total size of loaded sections: 2460
26481 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26528 [I][Elf] Total size of loaded sections: 988
26531 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26580 [I][Elf] Total size of loaded sections: 2044
26583 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26629 [I][Elf] Total size of loaded sections: 1464
26632 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26680 [I][Elf] Total size of loaded sections: 636
26682 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26737 [I][Elf] Total size of loaded sections: 1024
26740 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
26744 [D][NfcSupportedCards] Loaded 19 plugins
26759 [D][Iso14443_4aPoller] Read ATS success
26768 [D][MfDesfirePoller] Read version success
26771 [D][MfDesfirePoller] Read free memory success
26775 [D][MfDesfirePoller] Read master key settings success
26780 [D][MfDesfirePoller] Read master key version success
26785 [D][MfDesfirePoller] Read application ids success
26787 [D][MfDesfirePoller] Selecting app 0
26792 [D][MfDesfirePoller] Reading app 0
26803 [D][MfDesfirePoller] Can't read file 0 data without authentication
26807 [D][MfDesfirePoller] Selecting app 1
26811 [D][MfDesfirePoller] Reading app 1
26827 [D][MfDesfirePoller] Can't read file 1 data without authentication
26831 [D][MfDesfirePoller] Selecting app 2
26835 [D][MfDesfirePoller] Reading app 2
```

doomwastaken, 2024-07-22:
Judging by your screenshot you are not using official firmware. Try flashing latest release and verify if issue persists

closed 2024-07-22 by skotopes

s1rleblanc, 2024-07-31:
> Judging by your screenshot you are not using official firmware. Try flashing latest release and verify if issue persists

Hi!.. thank you for answer me.. it is OFW.

skotopes, 2024-07-31:
@s1rleblanc OFW means Official FirmWare(this repository), your screenshot is clearly indicating that you are using something different.

s1rleblanc, 2024-07-31:
> @s1rleblanc OFW means Official FirmWare(this repository), your screenshot is clearly indicating that you are using something different.

yeah.. i know what it means.. in OFW do the same thing.. that i want said.

skotopes, 2024-07-31:
Which specific build of OFW you were using? Reproduce this issue with latest release 0.104 and post here logs and backtrace(if you have debugger)

s1rleblanc, 2024-07-31:
> Which specific build of OFW you were using? Reproduce this issue with latest release 0.104 and post here logs and backtrace(if you have debugger)

ok.. i will post screenshots in a while.