flipperdevices/flipperzero-firmware #3820: subghz tx_from_file in protobuf API

gaellalire, 2024-08-01:
### Description of the feature you're suggesting.

In https://github.com/flipperdevices/flipperzero-protobuf
So that it can be called with Bluetooth LE.
If it is not wanted in protobuf, please provide a way to call it via Bluetooth.

gaellalire, 2024-08-04:
It can also be done by adding args to int32_t subghz_app(void* p).
In addition to subghz_app(args = "path") we can have subghz_app(args = "path <repeat:int> <autoclose/no-gui:bool>")

gaellalire, 2024-08-06:
So Sub-Ghz have a RPC mode, you need to launch with "RPC" arg in protobuf client side, the firmware will add the pointer to the RPC context needed by the Sub-Ghz app.
After that AppLoadFileRequest call will work, and AppButtonPressRequest will starts to emit the signal.

However AppButtonReleaseRequest is not working, even if there is some code about it.
The led keeps blinking indefinitely until AppExitRequest is sent.

You need to fix AppButtonReleaseRequest. AppButtonPressRequest args could contain a number of repeat to emit, in that case AppButtonReleaseRequest will only be usefull to abort. An event with 0 command id can be send when the message is fully repeated (as it is done for StartRequest / AppExitRequest).

Also if the RPC session is closed, Sub-Ghz app in RPC mode will crash instead of just close itself.

Skorpionm, 2024-08-14:
https://github.com/flipperdevices/flipperzero-firmware/pull/3838
Check

gaellalire, 2024-08-15:
Hello, fixing AppButtonReleaseRequest is just one part of this ticket.
I can create 3 new tickets if you want:
- tx_from_file in protobuf -> what I asked first, best for me because no need to start an app, need probably 2 new messages in protobuf (one for tx_from_file, one for cancel_tx_from_file, and one event [0 id] to notify that transmission is done)
- add repeat and autoclose args in subghz_app -> simple for you, but we need to launch the app and wait for the app to close on each message
- add repeat in AppButtonPressRequest for subghz_app -> the app can be kept opened, when the repeat is done an event [0 id] should be sent

Anyway we need a way to repeat a message, and be notified when 'repeat' messages is transmitted.

Skorpionm, 2024-08-17:
1. will not work. since protobuf does not know how to work with CLI applications. 2,3 sending data to SubGhz repeats the behavior of a real remote control, sends as much as the button is held down. if it is more convenient for you to use SubGhz tx_from_file with the required number of repetitions and other things, use the CLI directly without protobuf

gaellalire, 2024-08-17:
Hello @Skorpionm,

I don't understand you. To make 1 work you can introduce SendTxFromFile and AbortSendTxFromFile messages in protobuf. This will be handled the same way tx_from_file work (I guess by calling FuriHal method directly without launching SubGhz application). 
You can in the session context keep a pointer to what is currently sending so that abort will be possible, I think it is already the case for tx_from_file.

By CLI I guess you mean the USB CLI (before start_rpc_session is sent), my issue is that these are not callable though Bluetooth LE.

About 2 and 3, I don't know what is the expected behavior for AppButtonPressRequest but I recorded that the message is sent only once, maybe it is a bug and it is supposed to repeat indefinitely. In the case only one message is expected then you may allow to put 0 in protobuf arg so that it repeat indefinitely.

gaellalire, 2024-08-18:
> 1. 2,3 sending data to SubGhz repeats the behavior of a real remote control, sends as much as the button is held down

It means that if the message lasts 2 seconds but you press the button for only 1 second, the message will be cut in half ? That is a terrible behavior. I think when you press and release it should send the full message at least once. The behavior after the first message is sent (cutting the message) is OK.

Now I understand why you did not introduce an event to notify that the message is sent. It is because you don't look at the end of the message, you just wait for AppButtonReleaseRequest.

On the client side it is very bad to have to wait for a time before sending AppButtonReleaseRequest, because we cannot have the exact message time, we need to take account of the delay of the event transmission through BLE, and this delay could vary. For the moment I'm waiting 500ms which it is way too long for my message.

Skorpionm, 2024-08-18:
in RC systems, there is no feedback. and the transmitter does not understand whether the receiver has received the signal or not. also, at one frequency or another, not one system can work, but several, thereby spoiling each other's signals. 1 repetition of the message (the one that does not contain a preamble) is not enough for the receiver to adjust the AGC. I mean that even transmitting a full message in 99% of cases will not be enough for confident reception of the signal by the receiver. Therefore, it is done as in real devices, hold the button until it "opens"

gaellalire, 2024-08-18:
Yes for a human to hold until it opens is correct. But a program should be able to choose the number of repetition to send. I understand that the message may not be correctly sent (interferences ...), but by sent I mean sent to CC1101, not to the RF receiver.

I expect as a human to see how many time my message is sent while I hold the button.
For a program, a protobuf notification each time a message is sent, will allow to stop the emission when the program think there is enough repetition.

I was also surprised when I created a 2 seconds message and when I pressed and released the button, the transmission stopped. It was not intuitive for me.

gaellalire, 2024-09-10:
Great new version. I'm now able to send more than 1 message each second (previously 1 message each 2 or 3 seconds).
I will create other bugs for robustness, speed and control.