flipperdevices/flipperzero-firmware #3821: RFID Read problem

domthomas-dev, 2024-08-03:
### Description of the feature you're suggesting.

Hello, I was unable to read to simulate this RFID key. I attach here the RAW files which were read without problems. Can we do something to make this work?

### Anything else?

[RfidRecord.zip](https://github.com/user-attachments/files/16482714/RfidRecord.zip)
[image]

skotopes, 2024-08-12:
RAW data does looks meaningful.
Do you have any other info about this system? Product? Vendor?

domthomas-dev, 2024-08-16:
Here the picture of the reader mutancode clemsa if that can help you 
[image]

[image]

domthomas-dev, 2024-08-26:
Do you need something else ?

zinongli, 2024-09-05:
The reader is probably this one: https://dopromatic.com/ck-40/
The website for reader datasheets: https://clemsa.es/descargas-dispositivos-de-control/
This reader's datasheet: https://clemsa.es/wp-content/uploads/2024/03/CK-40-FICHA-TECNICA.pdf
It says it uses TK 40 tags, which probably means TK4100, quite like EM4100. 
TK4100 datasheet: https://www.egrfid.com/wp-content/uploads/2014/03/TK4100.pdf
This tag should've been readable by the EM4100 protocol. But it **seriously** violated the parity bits, row and column. Here's the most robust decoding I got from the raw files.

```
paragraph 0
preamble
11111111

data parity
0100 0
0100 0
0001 0
1010 0
0001 0
1101 1
0100 1
0100 0
0000 0
0000 0

paragraph 1
preamble
11111111

data parity
0100 0
0010 0
0001 0
0110 0
0000 1
1011 0
1100 0
1100 0
0000 0
000000 (yes six zeros)
```

Also this is quite a case like electra where there are two paragraphs of EM4100-variant paragraphs. Same case for Jantek tags who contains one parity-OK paragraph and another parity-violated paragraph to follow. Jantek tags are, in fact, T5577s though. 

I can dig more into the format and encoding. In the meantime, @domthomas-dev do you have a proxmark3?

domthomas-dev, 2024-09-08:
Is it possible to right an other card with it ?

domthomas-dev, 2024-10-06:
Is there any progress on this issue?

I dont understand what should i do with T5578

domthomas-dev, 2025-02-02:
Can you help me to copy this on an other card ?

zinongli, 2025-02-02:
Hi @domthomas-dev , as I have written in my reply, you can write this tag onto a T5577 by using the app and the file in the zip, both of which I have hyperlinked. If you click them you would see there are instructions in the app's read me telling you exactly how to write it to a T5577. Have you read it?