flipperdevices/flipperzero-firmware #3825: C++ Breakpoints not hit

janwiesemann, 2024-08-07:
### Describe the bug.

I'm currently trying to develop a small application for FZ. It utilizes some C++ Code and I would like to debug it. Sadly, it does not work. FreeRTOS and hardware close development is relatively new for me. I'm able to add C breakpoints. Yes, I could definitely write it in C but where is the fun in that?

Any help or hint would be appreciated.

### Reproduction

1. Create a new app
2. Add some C++ code
3. build it and try to attach a debugger to the application

### Target

f7 with VS-Code and C++

### Logs

```Text
### Debug Console

Cortex-Debug: VSCode debugger extension version 1.12.1 git(652d042). Usage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from /Users/user/Projects/flipper/flipperzero-firmware/toolchain/current/bin/arm-none-eabi-objdump --syms -C -h -w /Users/user/Projects/flipper/flipperzero-firmware/build/latest/firmware.elf
Reading symbols from /Users/user/Projects/flipper/flipperzero-firmware/toolchain/current/bin/arm-none-eabi-nm --defined-only -S -l -C -p /Users/user/Projects/flipper/flipperzero-firmware/build/latest/firmware.elf
Launching GDB: /Users/user/Projects/flipper/flipperzero-firmware/toolchain/current/bin/arm-none-eabi-gdb-py3 -q --interpreter=mi2
    IMPORTANT: Set "showDevDebugOutput": "raw" in "launch.json" to see verbose GDB transactions here. Very helpful to debug issues or report problems
Launching gdb-server: JLinkGDBServer -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32WB55RG -rtos GDBServer/RTOSPlugin_FreeRTOS.dylib
    Please check TERMINAL tab (gdb-server) for output from JLinkGDBServer
Finished reading symbols from objdump: Time: 42 ms
Output radix now set to decimal 10, hex a, octal 12.
Input radix now set to decimal 10, hex a, octal 12.
Finished reading symbols from nm: Time: 767 ms
vPortSuppressTicksAndSleep (expected_idle_ticks=30) at targets/f7/furi_hal/furi_hal_os.c:172
172	        return;
Program stopped, probably due to a reset and/or halt issued by debugger
Firmware version on attached Flipper:
	Version:     0.104.0
	Built on:    24-07-2024
	Git branch:  release
	Git commit:  ee1b8b9f
	Dirty:       False
	HW Target:   7
	Origin:      Official
	Git origin:  git@github.com:user/flipperzero-firmware.git,https://github.com/flipperdevices/flipperzero-firmware.git
Support for Flipper external apps debug is loaded
Set 'build/latest/.extapps' as debug info lookup path for Flipper external apps
Attaching to Flipper firmware
[New Remote target]
[New Thread 536882256]
[New Thread 537080456]
[New Thread 537081676]
[New Thread 536882912]
[New Thread 537106944]
[New Thread 537072504]
[New Thread 537085728]
[New Thread 537083920]
[New Thread 537085336]
[New Thread 536950776]
[New Thread 536886176]
[New Thread 537073724]
[New Thread 536910936]
[New Thread 536909416]
[New Thread 536884472]
[New Thread 537085532]
[New Thread 537074944]
[New Thread 536890328]
[New Thread 537084116]
[New Thread 537079236]

Thread
[... 46 lines cut ...]

### Setup:
- JLink Mini EDU
- VS-Code
- fbt (application in `applications_user/`; not using ufbt)

### Full Build log
```
rm -rf build/f7-firmware-C/.extapps/ && ./fbt COMPACT=1 DEBUG=0 VERBOSE=1 build APPSRC=applications_user/dolphin-jump/main.cpp

arm-none-eabi-g++ -o build/f7-firmware-C/.extapps/dolphin_jump/Game.o -c -std=c++20 -fno-rtti -fno-use-cxa-atexit -fno-exceptions -fno-threadsafe-statics -ftemplate-depth=4096 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -Wall -Wextra -Werror -Wno-error=deprecated-declarations -Wno-address-of-packed-member -Wredundant-decls -Wdouble-promotion -fdata-sections -ffunction-sections -fsingle-precision-constant -fno-math-errno -g -Os -mword-relocations -mlong-calls -fno-common -nostdlib -D_GNU_SOURCE -DFW_CFG_default -D"M_MEMORY_FULL(x)=abort()" -DSTM32WB -DSTM32WB55xx -DUSE_FULL_ASSERT -DUSE_FULL_LL_DRIVER -DMBEDTLS_CONFIG_FILE=\"mbedtls_cfg.h\" -DPB_ENABLE_MALLOC -DFW_ORIGIN_Official -DFURI_NDEBUG -DNDEBUG -DFAP_VERSION=\"0.1\" -Ifuri -Iapplications/services -Iapplications/main -Itargets/furi_hal_include -Itargets/f7/ble_glue -Itargets/f7/fatfs -Itargets/f7/furi_hal -Itargets/f7/inc -I. -Ilib -Ibuild/f7-firmware-C/assets/compiled -Ilib/mlib -Ilib/cmsis_core -Ilib/stm32wb_cmsis/Include -Ilib/stm32wb_hal/Inc -Ilib/stm32wb_copro/wpan -Ilib/drivers -Ilib/FreeRTOS-Kernel/include -Ilib/FreeRTOS-Kernel/portable/GCC/ARM_CM4F -Ilib/FreeRTOS-glue -Ilib/microtar/src -Ilib/mbedtls/include -Ilib/toolbox -Ilib/libusb_stm32/inc -Ilib/drivers -Ilib/flipper_format -Ilib/one_wire -Ilib/ibutton -Ilib/infrared/encoder_decoder -Ilib/infrared/worker -Ilib/littlefs -Ilib/subghz -Ilib/nfc -Ilib/digital_signal -Ilib/pulse_reader -Ilib/signal_reader -Ilib/app-scened-template -Ilib/callback-connector -Ilib/u8g2 -Ilib/lfrfid -Ilib/flipper_application -Ilib/music_worker -Ilib/mjs -Ilib/nanopb -Ilib/heatshrink -Ilib/ble_profile -Ilib/bit_lib -Ilib/datetime -Ibuild/f7-firmware-C/.extapps/dolphin_jump -Iapplications_user/dolphin-jump applications_user/dolphin-jump/Game.cpp
python3 scripts/assets.py icons applications_user/dolphin-jump/images build/f7-firmware-C/.extapps/dolphin_jump --filename dolphin_jump_icons
arm-none-eabi-g++ -o build/f7-firmware-C/.extapps/dolphin_jump/main.o -c -std=c++20 -fno-rtti -fno-use-cxa-atexit -fno-exceptions -fno-threadsafe-statics -ftemplate-depth=4096 -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -Wall -Wextra -Werror -Wno-error=deprecated-declarations -Wno-address-of-packed-member -Wredundant-decls -Wdouble-promotion -fdata-sections -ffunction-sections -fsingle-precision-constant -fno-math-errno -g -Os -mword-relocations -mlong-calls -fno-common -nostdlib -D_GNU_SOURCE -DFW_CFG_default -D"M_MEMORY_FULL(x)=abort()" -DSTM32WB -DSTM32WB55xx -DUSE_FULL_ASSERT -DUSE_FULL_LL_DRIVER -DMBEDTLS_CONFIG_FILE=\"mbedtls_cfg.h\" -DPB_ENABLE_MALLOC -DFW_ORIGIN_Official -DFURI_NDEBUG -DNDEBUG -DFAP_VERSION=\"0.1\" -Ifuri -Iapplications/services -Iapplications/main -Itargets/furi_hal_include -Itargets/f7/ble_glue -Itargets/f7/fatfs -Itargets/f7/furi_hal -Itargets/f7/inc -I. -Ilib -Ibuild/f7-firmware-C/assets/compiled -Ilib/mlib -Ilib/cmsis_core -Ilib/stm32wb_cmsis/Include -Ilib/stm32wb_hal/Inc -Ilib/stm32wb_copro/wpan -Ilib/drivers -Ilib/FreeRTOS-Kernel/include -Ilib/FreeRTOS-Kernel/portable/GCC/ARM_CM4F -Ilib/FreeRTOS-glue -Ilib/microtar/src -Ilib/mbedtls/include -Ilib/toolbox -Ilib/libusb_stm32/inc -Ilib/drivers -Ilib/flipper_format -Ilib/one_wire -Ilib/ibutton -Ilib/infrared/encoder_decoder -Ilib/infrared/worker -Ilib/littlefs -Ilib/subghz -Ilib/nfc -Ilib/digital_signal -Ilib/pulse_reader -Ilib/signal_reader -Ilib/app-scened-template -Ilib/callback-connector -Ilib/u8g2 -Ilib/lfrfid -Ilib/flipper_application -Ilib/music_worker -Ilib/mjs -Ilib/nanopb -Ilib/heatshrink -Ilib/ble_profile -Ilib/bit_lib -Ilib/datetime -Ibuild/f7-firmware-C/.extapps/dolphin_jump -Iapplications_user/dolphin-jump applications_user/dolphin-jump/main.cpp
_validate_api_cache(["targets/f7/api_symbols.csv"], ["build/f7-firmware-C/sdk_origin.i"])
arm-none-eabi-gcc -o build/f7-firmware-C/.extapps/dolphin_jump/dolphin_jump.o -c -std=gnu2x -Wstrict-prototypes -Wno-strict-prototypes -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -Wall -Wextra -Werror -Wno-error=deprecated-declarations -Wno-address-of-packed-member -Wredundant-decls -Wdouble-promotion -fdata-sections -ffunction-sections -fsingle-precision-constant -fno-math-errno -g -Os -mword-relocations -mlong-calls -fno-common -nostdlib -D_GNU_SOURCE -DFW_CFG_default -D"M_MEMORY_FULL(x)=abort()" -DSTM32WB -DSTM32WB55xx -DUSE_FULL_ASSERT -DUSE_FULL_LL_DRIVER -DMBEDTLS_CONFIG_FILE=\"mbedtls_cfg.h\" -DPB_ENABLE_MALLOC -DFW_ORIGIN_Official -DFURI_NDEBUG -DNDEBUG -DFAP_VERSION=\"0.1\" -Ifuri -Iapplications/services -Iapplications/main -Itargets/furi_hal_include -Itargets/f7/ble_glue -Itargets/f7/fatfs -Itargets/f7/furi_hal -Itargets/f7/inc -I. -Ilib -Ibuild/f7-firmware-C/assets/compiled -Ilib/mlib -Ilib/cmsis_core -Ilib/stm32wb_cmsis/Include -Ilib/stm32wb_hal/Inc -Ilib/stm32wb_copro/wpan -Ilib/drivers -Ilib/FreeRTOS-Kernel/include -Ilib/FreeRTOS-Kernel/portable/GCC/ARM_CM4F -Ilib/FreeRTOS-glue -Ilib/microtar/src -Ilib/mbedtls/include -Ilib/toolbox -Ilib/libusb_stm32/inc -Ilib/drivers -Ilib/flipper_format -Ilib/one_wire -Ilib/ibutton -Ilib/infrared/encoder_decoder -Ilib/infrared/worker -Ilib/littlefs -Ilib/subghz -Ilib/nfc -Ilib/digital_signal -Ilib/pulse_reader -Ilib/signal_reader -Ilib/app-scened-template -Ilib/callb [... cut ...]
arm-none-eabi-objcopy --remove-section .ARM.attributes --add-section .fapmeta=build/f7-firmware-C/.extapps/dolphin_jump/.fapmeta --set-section-flags .fapmeta=contents,noload,readonly,data --strip-debug --strip-unneeded --add-gnu-debuglink=build/f7-firmware-C/.extapps/dolphin_jump_d.elf build/f7-firmware-C/.extapps/dolphin_jump_d.elf build/f7-firmware-C/.extapps/dolphin_jump.fap
python3 scripts/fastfap.py build/f7-firmware-C/.extapps/dolphin_jump.fap arm-none-eabi-objcopy
_validate_app_imports(["build/f7-firmware-C/.extapps/dolphin_jump.impsyms"], ["build/f7-firmware-C/.extapps/dolphin_jump.fap"])
```

skotopes, 2024-08-12:
To be honest that looks like issue with ISP. 

@hedger @DrZlo13 FYI

hedger, 2024-08-12:
@janwiesemann please provide source code for reproducing the issue.

janwiesemann, 2024-08-12:
@hedger The example code can be found at https://github.com/janwiesemann/flipperzero-cpp-debugging-issue-example

I think a lot of issued might be caused by the `-Os` flag.

janwiesemann, 2024-08-20:
@hedger I've just testet it with the WiFi Dev Board and backmagic. The same issue is present. My best guess is, that the optimization is messing with something and breaking some debugging stuff.

EstebanFuentealba, 2024-08-28:
> @hedger The example code can be found at https://github.com/janwiesemann/flipperzero-cpp-debugging-issue-example
> 
> I think a lot of issued might be caused by the `-Os` flag.

I [also encountered different results](https://github.com/flipperdevices/flipperzero-firmware/issues/3531) when using ufbt and fbt. The issue was that ufbt applies the -Os optimization flag. My quick solution at the time was to use the following directive:

```c
#pragma GCC optimize("O1")
```