flipperdevices/flipperzero-firmware #3832: Noise recording in "Read raw" despite high RSSI settings

OevreFlataeker, 2024-08-09:
### Describe the bug.

Observation:
I noticed that when I try to record a transmission from a key fob the resulting SUB file is extremely noise with lots of long "high" sequences when looking at the file in the pulse plotter.

This wouldn´t be surprising at all if I would not have filtered using RSSI, but I have set the RSSI filter to -50 in order to avoid all this background noise. 

Expectation:

The resulting SUB file should only contain those signal parts which surpass the RSSI -50 threshold and not the noise floor. 

Reproduction:

Open SubGhz App->Read Raw. Configure the RSSI threshold to be -50. Start the recording. Verify the noise floor is way below the threshold line. Press a button on the sender. Observe the pulses above the threshold. Save the SUB file. Inspect in Pulse plotter.

Also: Replay the recording and record and visualize with URH. Of course the same noisy signal is there.

Configure recording params:
[image]

Record something:
[image]

Visualize:
[image]

URH Recording of Flipper's playback of the recording:
[image]

While it still MIGHT be able to decode the intended signal from that raw recording

```
>: subghz decode_raw /ext/subghz/Noise.sub
Load_keystore keeloq_mfcodes OK
Load_keystore keeloq_mfcodes_user Absent
Listening at /ext/subghz/Noise.sub.

Press CTRL+C to stop

Packets received 0

>: subghz decode_raw /ext/subghz/Noise270.sub
Load_keystore keeloq_mfcodes OK
Load_keystore keeloq_mfcodes_user Absent
Listening at /ext/subghz/Noise270.sub.

Press CTRL+C to stop

Dickert_MAHS
User-Dips:      +00-+---0+
Fac-Code:       0000-000
Dickert_MAHS
User-Dips:      +00-+---0+
Fac-Code:       0000-000

Packets received 2
```
doing a raw recording in order to initially analyze the observed signal and find it's structure etc. is impossible with those recordings (because you can´t tell it apart from the noise present).

Questions are:
  * Is it expected to have such a noisy recording despite the high RSSI filter? What's the point of the filter then?
  * Is the Read RAW mode actually meant for recordings to initially analyze an unknown signal? How should it be done given the noisy recording? 
  * Looking at the SUB file it's actually more "high" than "low" signal parts - in contrast to what's seen in the flipper's display. How can that be explained? It would at least expect some long "low" passages matching the durations I see in the display where no signal surpasses the RSSI threshold barrier. 
  

[ExampleRecordings.zip](https://github.com/user-attachments/files/16559804/ExampleRecordings.zip)

### Reproduction

1. Open SubGhz App->Read Raw. 
2. Configure the RSSI threshold to be -50. 
3. Start the recording. 
4. Verify the noise floor is way below the threshold line. 
5. Press a button on the sender couple of times. 
6. Observe only those pulses are above the threshold. 
7. Save the SUB file. 
8. Inspect in Pulse plotter.

### Target

SUB file only contains those parts which are above the RSSI filter

### Logs

```Text
Please see attached SUB files above.
```

doomwastaken, 2024-08-09:
@OevreFlataeker greetings. What firmware version did you use?

OevreFlataeker, 2024-08-09:
Oh I am sorry, that's an important point I missed 😂

I am on latest 0.104.0 but it was the same behavior for me on at least the previous version as well.

Am I using the Read Raw feature in a wrong way?

Skorpionm, 2024-08-09:
Rssi is not a filter, but simply a sign that there is something on the air, the recording continues for about 2 seconds after the last Rssi above the threshold was registered

Skorpionm, 2024-08-09:
Read RAW writes everything it hears, without understanding what exactly, like a voice recorder

OevreFlataeker, 2024-08-09:
I've made some more tests, this time with RSSI set to -50 in the settings and long presses on the button.

[RAW-20240809-223343.sub.zip](https://github.com/user-attachments/files/16566729/RAW-20240809-223343.sub.zip)

This looks like a very good recording to me. 

Zooming in on one of the "solid blocks" when fully zommed out gives:

[image]

That looks totally fine, and also the statistical distribution of pulses and gaps is plausible: Very high weight on length of around 400 and 800.

Still I don´t really get why my initial recording is so crap. Was it just a bad moment to record you think?

OevreFlataeker, 2024-08-09:
> Rssi is not a filter, but simply a sign that there is something on the air, the recording continues for about 2 seconds after the last Rssi above the threshold was registered

So it's just the _trigger_ for the recording to start? And until the recording stops (until 2 seconds after RSSI drops enough) _EVERYTHING_ which can be received is recorded? Not only signals above the RSSI threshold value? So the noise floor and my true signal will be equally be present in the recording during that time? Is this what you're saying? Or did I misunderstand this?

What confuses me is that I understood when doing a recording with a SDR, the RSSI value can actually be used as a filter not just a trigger. But is this then the fundamental difference between a SDR and the Flipper, because the Flipper has only a concept of either "high" or "low" but not about "strength of high"?

Skorpionm, 2024-08-10:
If the Cc1101 could work as an SDR, many compromises could be avoided

closed 2024-08-12 by skotopes