flipperdevices/flipperzero-firmware #3833: JS - Token reading

Tomi2965, 2024-08-09:
### Describe the enhancement you're suggesting.

It's possible, to add a commands for JavaScript for reading/get tokens ID's? I mean, any. SubGHz, RFID, NFC, ... . 

This function will help me and my colleagues to make  our work much & much faster.  In my case - use it to scan all new orders of tokens to system (NFC, RFID, iButton, for now, we are typing all ID's by hand, which cause mistakes from time to time). Then JS just send ID as a keyboard to PC. Of course, I will need to use JS to transform ID to different format. Manufacturer FDI write ID like 01AB34CD, but Flipper read it as CD34AB01. 

Of course, I was try to search for possible commands in Flipper's JS, if these functions are implemented, then please sorry (send me a link). If not, it's possible to make a solution? Flipper with JS but without ability to return token ID is just a waste of anything. But.. If Flipper can do this, it will make Flipper more customizable and usable.

Please, leave any comments, ask any questions. 

PS: big thanks to anyone, who constructively helps to make Flipper better ♥️

Tomi2965, 2025-02-19:
@portasynthinca3 - is there any progress with this idea? Thanks a lot! :)

portasynthinca3, 2025-03-05:
@Tomi2965 unfortunately, the "then return all data to JS" is exactly the hard problem here. Every card standard requires its own data representation, which in turn requires us to write unique converts between the C representation and the JS representation. And that's not the only roadblock.

However, I have some good news for you! I just learned that one of my colleagues (who's way more well versed in NFC and RFID than I am) is currently actively working on a Proxmark-like CLI, which looks like it might fit your particular use case of dumping scanned UIDs over USB.