flipperdevices/flipperzero-firmware #3835: Attempting to read DESFire Card causes `furi_check()` crash

5aji, 2024-08-11:
### Describe the bug.

Reading a DESFire card crashes the Flipper with a `furi_check()` error. It seems to crash after trying to read the second block. This is a card I had lying around, and I do not possess the keys (nor am I trying to get them). This happens on both 0.104.0 and 0.105.0-RC.

### Reproduction

1. Open NFC App -> Read
2. Alternatively, select Extra-> Read Specific -> DESFire
3. Touch Flipper to keycard
4. Lights blink for a split second and then the system reboots.

### Target

Mifare DESFire Card

### Logs

```Text
1050914 [D][NfcSupportedCards] Loaded 19 plugins
1050929 [D][Iso14443_4aPoller] Read ATS success
1050938 [D][MfDesfirePoller] Read version success
1050942 [D][MfDesfirePoller] Read free memory success
1050946 [D][MfDesfirePoller] Read master key settings success
1050950 [D][MfDesfirePoller] Read master key version success
1050955 [D][MfDesfirePoller] Read application ids success
1050958 [D][MfDesfirePoller] Selecting app 0
1050963 [D][MfDesfirePoller] Reading app 0
1050974 [D][MfDesfirePoller] Can't read file 0 data without authentication
1050978 [D][MfDesfirePoller] Selecting app 1
1050983 [D][MfDesfirePoller] Reading app 1
1050999 [D][MfDesfirePoller] Can't read file 1 data without authentication
1051002 [D][MfDesfirePoller] Selecting app 2
1051007 [D][MfDesfirePoller] Reading app 2

[CRASH][NfcWorker] furi_check failed
        r0 : 20025fe8
        r1 : 200251d0
        r2 : 0
        r3 : 0
        r4 : 0
        r5 : 0
        r6 : 20025fe8
        r7 : 200251d0
        r8 : 80a22e7
        r9 : 80a23d6
        r10 : 80a23e8
        r11 : 20031364
        lr : 80389d9
        stack watermark: 7556
             heap total: 186064
              heap free: 31576
         heap watermark: 27568
        core2: not faulted
Rebooting system�0
```

### Anything else?

I have a Proxmark3 if that would help provide more information.

skotopes, 2024-08-12:
@5aji any details on the card itself? Vendor? Type? if you have promark can you also post details scan details from it?

5aji, 2024-08-12:
The key is a Walt Disney World keycard. 
<details><summary>Image of back of card</summary>
<p>

There are no markings on the front.

[image]

</p>
</details> 

<details><summary>Proxmark info</summary>
<p>

```
[usb] pm3 --> hf 14a info

[+]  UID: 04 3A 34 4A 52 67 80 
[+] ATQA: 03 44
[+]  SAK: 20 [1]
[+] MANUFACTURER: NXP Semiconductors Germany
[+] Possible types:
[+]    MIFARE DESFire CL2
[+]    MIFARE DESFire EV1 256B/2K/4K/8K CL2
[+]    MIFARE DESFire EV2 2K/4K/8K/16K/32K
[+]    MIFARE DESFire EV3 2K/4K/8K
[+]    MIFARE DESFire Light 640B
[+]    NTAG 4xx
[=] -------------------------- ATS --------------------------
[+] ATS: 06 75 77 81 02 80 [ F0 00 ]
[=]      06...............  TL    length is 6 bytes
[=]         75............  T0    TA1 is present, TB1 is present, TC1 is present, FSCI is 5 (FSC = 64)
[=]            77.........  TA1   different divisors are supported, DR: [2, 4, 8], DS: [2, 4, 8]
[=]               81......  TB1   SFGI = 1 (SFGT = 8192/fc), FWI = 8 (FWT = 1048576/fc)
[=]                  02...  TC1   NAD is NOT supported, CID is supported

[=] -------------------- Historical bytes --------------------
[+] 80  (compact TLV data object)

[usb] pm3 --> hf mfdes info

[=] ---------------------------------- Tag Information ----------------------------------
[+]               UID: 04 3A 34 4A 52 67 80 
[+]      Batch number: B9 0C 10 45 60 
[+]   Production date: week 16 / 2020

[=] --- Hardware Information
[=]    raw: 04010201001205
[=]      Vendor Id: NXP Semiconductors Germany
[=]           Type: 0x01 ( DESFire )
[=]        Subtype: 0x02
[=]        Version: 1.0 ( DESFire EV1 )
[=]   Storage size: 0x12 ( 512 bytes )
[=]       Protocol: 0x05 ( ISO 14443-2, 14443-3 )

[=] --- Software Information
[=]    raw: 04010101051205
[=]      Vendor Id: NXP Semiconductors Germany
[=]           Type: 0x01 ( DESFire )
[=]        Subtype: 0x01
[=]        Version: 1.5
[=]   Storage size: 0x12 ( 512 bytes )
[=]       Protocol: 0x05 ( ISO 14443-3, 14443-4 )

[=] --------------------------------- Card capabilities ---------------------------------

[+] --- AID list
[+] AIDs:  f70090, 78e127, 4c4344

[+] ------------------------------------ PICC level -------------------------------------
[+] Applications count: 3 free memory 128 bytes
[+] PICC level auth commands: 
[+]    Auth.............. NO
[+]    Auth ISO.......... NO
[+]    Auth AES.......... YES
[+]    Auth Ev2.......... NO
[+]    Auth ISO Native... YES
[+]    Auth LRP.......... NO
[+] PICC level rights:
[+] [1...] CMK Configuration changeable   : YES
[+] [.1..] CMK required for create/delete : NO
[+] [..1.] Directory list access with CMK : NO
[+] [...1] CMK is changeable              : YES
[+] 
[+] Key: 2TDEA
[+] key count: 1
[+] PICC key 0 version: 254 (0xfe)

[=] --- Free memory
[+]    Available free memory on card         : 128 bytes

[=] Standalone DESFire
```
</p>
</details> 

If necessary I can pull out my debugger and try and find exactly what check fails.

skotopes, 2024-08-12:
If you can get backtrace from debugger that will simplify everything

gzub, 2025-04-23:
I ran into the same problem with a Disney DESFire Card, failing on the same application, causing the flipper zero to reboot and came across this while researching the issue...  The back of the card looks identical to the image...

closed 2026-08-17 by turbospok