flipperdevices/flipperzero-firmware #3844: Sub-Ghz transmission is skipping some bits

gaellalire, 2024-08-16:
### Describe the bug.

I'm trying to send a message with Sub-Ghz app with a 25µs bit.
I tried both Raw and BinRaw format, with Raw a bit is lost every 25 to 40 bits, with BinRaw a bit is lost every 22 to 26 bits.

PandwaRF (CC1111) is able to reproduce correctly the message therefore I think that CC1101 should also be able to reproduce it. I think the issue is in the communication between FlipperZero processor and CC1101, maybe it is a hardware issue (too slow memory or processor) or maybe the communication can be improved in this firmware code.

[clock.zip](https://github.com/user-attachments/files/16636929/clock.zip)

### Reproduction

Unzip file copy to subghz, emit while recording with another device.
I added spaces for missing bits, so it aligns with expected message

```
-- binraw
# expected
1010101010011101010101001110101010100111010101010011101010101001110101010100111010101010011101010101001110101010100111010101010011101010101001110101010100111010101010011101010101001111111111111111010001111111100101011001110101010100111010101010011101010101001110101010100111010101010011101010011001110100110100111001101010011101010011001110010110100111010101010011101010100101110010101100111011110001111
# recorded
                                        00010101001110101010 001110101010100111010101 100111010101010011101010 01001110101010100111010 010100111010101010011101 10101001110101010100111111111111111101000111111110 10101100111010101010011 0101010100111010101010011 0101010100111010101010 1110101001100111010011010 11100110101001110101001 001110010110100111010100 10011101010100101110010 01100111011110001111 0
000010101001110101010100 1101010101001110101010100 1101010101001110101010100 11010101010011101010101 011101010101001110101010 001110101010100111010101 100111010101010011101010 010011111111111111111010011111111001010110011101 101010011101010101001110 010101001110101010100111 101010100111010100110011 0100110100111001101010011 0101001100111001011010 111010101010011101010100 011100101011001110111100 1111
     000000101010011101001010011101010101001110110101001110101010100111001010100111010101010011010101010011101010101001101010101001110101010100110101010100111010101010111010101010011111111111111110100011111111001010100111010101010011101010010011101010101001110101101001110101010100111011001100111010011010011101101010011101010011001110101101001110101010100110101010010111001010110011011110001111000000101000111010101010011101010110011101010101001110101101001110101010100111010010100111010101010011101101010011101010101001110010101001110101010100110101010100111010101010011111111111111110100011111110010101100111010101010111010101010011101010100011101010101001110101011001110101001100111010010100111001101010011101010110011100101101001110110101001110101010010111010101100111011110001111000001010100111010101010011010101010011101010101011101010101001110101010001110101010100111010101100111010101010011101010010011101010101001110100101001110101010100111011010100111111111111111110000111111110010101100110101010100111010101010011010101010011101010101001101010101001110101001101110100110100111001101000111010100110011100101110011101010101001110101000101110010101100111011100011110000001010100111101010100111010101010011010101010011101010101001101010101001110101010100110101010100111010101010111010101010011101010100011101010101001110101011001110101010100111111111111111101000111111110011011001110101010100111011010100111010101010011100101010011101010101001111010011001110100110100110011010100111010100110011001011010011101010101011101010100101110010101101110111100011110000001101001110101010100111010010100111010101010011101101010011101010101001111010101001110101010100110101010100111010101010011010101010011101010101001101010101001110101010101111111111111111101000111111100101011001110101011001110101010100111010100100111010101010011101001010011101010011001110101101001110011010100111001001100111001011010011110101010011101010100101100101011001110111100011100000010101001110101010001110101010100111010101100111010101010011101011010011101010101001110100101001110101010100111011010100111010101010011100101010011101010101001101010101001111111111111111010001111111100101011001101010101001110101010101110101010100111010101000111010101010011101010010011101001101001110011001001110101001100111001110100111010101010011101101001011100101011001110111000111100000010101001101010101001110101010101110101010100111010101010111010101010011101010100011101010101001110101011001110101010100111010110100111010101010011101001010011101010101001111111111111111010001111111101010110011101010101001101010101001110101010100110101010100111010101010011010100110011101001101011100110101001110101001001110010110100111010101100111010101001011100101110011101111000111100000101010011101010101001111010101001110101010100110101010100111010101010011010101010011101010101001101010101001110101010101110101010100111010101000111010101010011101010110011111111111111111010011111111001010110011101001010011101010101001110110101001110101010100111001010100111010100110011010011010011100110101001101010011001110010110100110101010100111010101001111001010110011101111001111000000101010011101011010011101010101001110100101001110101010100111011010100111010101010011110101010011101010101001101010101001110101010100110101010100111010101010011010101010011111111111111110100011111111001010100111010101010011101010110011101010101001110101001001110101010100111010

-- raw
 # expected
1010101010011101010101001110101010100111010101010011101010101001110101010100111010101010011101010101001110101010100111010101010011101010101001110101010100111010101010011101010101001111111111111111010001111111100101011001110101010100111010101010011101010101001110101010100111010101010011101010011001110100110100111001101010011101010011001110010110100111010101010011101010100101110010101100111011110001111
# recorded    10101010011101010000011101010101001110101010000110101010100111010101010011101001010011101010101001110101010001110101010100111010101010011010101010011101010101001110101000001111111111111111101000111111110010101100110 1010101001110101010100111010101010 11101010101001110101010100 11101010110011101001101001110011010100 11010100110011100101101001110101 10100111010101001011100101011001110 1110001111

```

rhradecky, 2024-08-16:
ds

Skorpionm, 2024-08-19:
[image]

and here is the signal that comes to the second CC1101. Your preset is not formed correctly. in general, if you want to transmit such short pulses and receive them correctly, consider the possibility of receiving the transmission in burst mode

closed 2024-08-19 by skotopes

gaellalire, 2024-08-19:
@Skorpionm,
The issue is clearly on transmission side. I'm using a LimeSDR at 3.2 M Sps (an IQ device), the signal is clean enough to be sure that some bits are skipped (also the consistency of 1 bit each ~25 bits).
Also tested with an external CC1101, it is worse (1 bit lost each ~15 bits).
I attached the IQ files if you want to check with URH yourself.

That being said how can you be sure that all bit are well emitted, did you debug the emission with Logic ?

It may be an issue with my Flipper. Also the preset is my sub files may be wrong. Did you use my sub files ? Which CC1101 field(s) should I change or add to improve the transmission ?

[FlipperIQ.zip](https://github.com/user-attachments/files/16662592/FlipperIQ.zip)

gaellalire, 2024-08-19:
If I ask for 50 µs bit duration, all the 50 messages are recognized however sometime an isolated bit can last only 26 µs (sometimes not enough).
With 40 µs bits duration only 3 of 50 messages are recognized, and same thing an isolated bit can last only 26 µs (sometimes not enough).
With 20 µs bits duration, an isolated bit can last between 24 µs and 26 µs (always too much) and all messages are wrong.

Skorpionm, 2024-08-20:
I debug the log analyzer signal that goes to the GO pin from Flipper to CC1101 in transmission mode, and comes from CC1101 in reception mode. Yes, I used your preset from your files in transmission mode and in reception mode. Data on the GO pin goes correctly, but what is transmitted on the air and received from it by CC1101 already directly depends on the filled preset. In ASK mode, the baundrate parameter affects the duration of pulse transmission; for 25us, it should be equal to 40 kbaud. Well, and some others to optimize reception. Use SmartRF Studio to create the correct preset

Skorpionm, 2024-08-20:
regarding the external module, using it in the data reception and transmission mode via G0, you will never receive odd durations, only even ones, i.e. 2, 4,... 24,26.... this is due to the peculiarity of the timer that services the external CC1101. there is no such difficulty when working with the internal CC1101

Skorpionm, 2024-08-20:
I'm very glad that you heard me. But I didn't understand about TE

gaellalire, 2024-08-20:
TE is the quantization interval in the key files, if it is set to a value below 50 µs then Sub-Ghz should set MDMCFG4 & MDMCFG3 to a fitting value.
If Sub-Ghz does that we don't need to put MDMCFG4 & MDMCFG3 in the preset. Like today we don't need to put the baseFrequency in the preset.

Raw files however don't have TE, but it can be calculated by looking at the minimum absolute value.

Skorpionm, 2024-08-21:
no it doesn't make sense. because when you receive you don't know what exactly you want to receive. if you want to receive send very short durations make a custom preset for this and use it both when transmitting and when receiving