flipperdevices/flipperzero-firmware #3845: Have the main IR app throw a custom exception for when the user attempts to open and use libary files directly

skyhawkillusions, 2024-08-18:
### Describe the enhancement you're suggesting.

**FEATURE REQUEST:** Change the error message that is thrown when attempting to open an IR library file to specifically indicate that the file is a library file, discourage the user from trying this again, and point to the proper usage of the library file in the universal remote functionality

**CONTEXT:** In multiple discord servers dealing with both stock firmware and any CFW forks directly utilizing this app unchanged, I have been observing complaints about the user being unable to load any of the files in [`/ext/infrared/assets/`](https://github.com/flipperdevices/flipperzero-firmware/tree/dev/applications/main/infrared/resources/infrared/assets). This occurs regardless of how the library file is opened, whether by the file browser, or by opening it through the IR app. Note that the IR app normally discourages opening the library files to begin with by hiding the `assets` folder; however, this can easily be overcome by (1) directly opening the library file from the in-device file browser, or (2) copying the library file outside the assets folder and into the open.

**PROBLEM:** The error message is in the form of `Failed to load "[filepath]"`. There is no way to tell from this message why the file failed to load, masking the problem and leaving the user in the dark regarding what is the next step. The immediate impression the user gets upon receiving this message is that something is wrong with the file.

**PROPOSED SOLUTION:** The addition of a conditional clause to the exception block thrown when the file cannot be properly loaded to change the message being returned in the event that the file turns out to be a library file. Wording could be something like "Library files can't be accessed normally, please use the universal remotes". A Boolean flag defaulting to `false` could be triggered if the header of the .ir file matches `Filetype: IR library file`, and could in turn redirect program flow to this special subclause. 

**EXAMPLE:** Three .ir files in a single folder called `defective.ir`, `library.ir`, and `signal.ir`. Their properties are as follows: 

- `defective.ir` is completely broken and cannot function in either the main IR app or the Universal remote app. 
- `library.ir` is properly functioning and can only be implemented by the universal remote functionality.
- `signal.ir` is properly functioning and can be implemented by the main IR app.

CURRENT BEHAVIOR:

- `defective.ir`: Throws the error `Failed to load "/any/infrared/defective.ir"`
- `library.ir`: Throws the error `Failed to load "/any/infrared/library.ir"`
- `signal.ir`: Proceeds without issue

DESIRED BEHAVIOR:

- `defective.ir`: Throws the error `Failed to load "/any/infrared/defective.ir"`
- `library.ir`: Throws the error `Library files can't be accessed normally, please use the universal remotes`
- `signal.ir`: Proceeds without issue

skotopes, 2024-09-13:
Fixed in https://github.com/flipperdevices/flipperzero-firmware/pull/3859

closed 2024-09-13 by skotopes

skyhawkillusions, 2024-09-14:
Thank fuck 👍👍👍👍👍👍👍👍👍👍