flipperdevices/flipperzero-firmware #3853: Reading an emulated Amiibo causes Flipper Zero to crash using NFC read but works using NFC Read specific card type

ukscone, 2024-08-24:
### Describe the bug.

When trying to read an emulated Amiibo from a device like the Flashiibo (https://www.flashiibo.com/) or Maleen NFC Tag Generator that uses an nRF52832 to emulate NTAG N215 tags the Flipper zero crashes and reboots with a furi_check failed error & the last line in the logs in debug or trace mode is "[D][NfcScanner] Found 5 children" when using the NFC Read but when using NFC->Extra Actions->Read Specific Card Type as either iso14443-3a or NTAG it works as expected.

### Reproduction

Apps->NFC->NFC->Read then hold a multi-ntag emulator device to the back of the flipper and watch the flipper screen darken and then "reboot" with a flipper crashed and was rebooted furi_check failed message

doomwastaken, 2024-08-27:
Greetings! We will get back to you, once we get the tag

ukscone, 2024-08-30:
Great. Thanks

doomwastaken, 2024-09-18:
Card had arrived, we will get back to you

doomwastaken, 2024-09-18:
@ukscone could you please verify what firmware version are you using? 
Additionally, does it persist on latest firmware?

ukscone, 2024-09-18:
i'm now on 1.0.1 firmware and yes it still crashes with a generic read and works when read as specific card type

doomwastaken, 2024-10-04:
Currently it is in backlog, thank you for reporting it