flipperdevices/flipperzero-firmware #3857: NTAG NFC Counter & ASCII Counter Mirror when emulating NTAG N2XX tags

ukscone, 2024-08-26:
### Describe the enhancement you're suggesting.

NTAG N2XX tags (I think other NXP NTAG versions have it too but I don't think the flipper emulates those tags) have something called NFC Counter and ASCII Counter Mirror see Table 11 & Sections 8.6 & 8.7 in https://www.nxp.com/docs/en/data-sheet/NTAG213_215_216.pdf. While I've only seen it used a few times in the wild mostly art projects & a couple of things i've done it would be nice if the Emulate function in the Flipper NFC app could replicate this with emulated NTAGs. I'm not sure it's doable as does the flipper know it's been read when it's in emulate mode but if it is it would be a nice added ability.

RebornedBrain, 2024-09-11:
Hi, @ukscone according to NXP specs NTAG21x tags and NTAG203 have a little bit different counter feature. 

### Counter feature
- **NTAG21x** tags have special READ_CNT command in order to read counter which is stored in some sort of shadow register and is not visible while dumping for example. And this counter increases every first READ or FAST_READ command. By default this counter is disabled and all attempts to read it will result NAK. Also there is a little bit lack of information about the fact, that in those NTAGs there is only one counter (which has number 2 counting from 0), while READ_CNT command supports up to 3 counters. This can lead to some misunderstanding about that counter mode and how it works.

- While on the other hand **NTAG203** has counter present on page 29h and which can be incremented by writing values directly to that page. 

### ASCII mirror
Only **NTAG21x** support it, not **NTAG203**.

### Flipper counter emulation
Flipper already can emulate counter for NTAG21X tags. You can try it by doing such things with the help of proxmark:

1. Read any clean NTAG213 with your Flipper and start emulation.
2. Run `hf mfu info` in order to see general card info, then in output find '**Tag configuration**' block, where counter status will be shown, by default it is disabled.
[image]

3. Then try to read counter with READ_CNT = 0x39 command, where second byte is counter number
`hf 14a raw -sc 3900` 
`hf 14a raw -sc 3901` 
`hf 14a raw -sc 3902` 
You will get such result, where 00 means NAK. 
[image]

4. Read config page by executing `hf mfu rdbl -b 42` the output sould be smth like that:
[image]

5. Now we modify first page byte and execute such command 'hf mfu wrbl -b 42 -d 10050000' where 0x10 means NFC_CNT
 _EN = 1 (see table 10 on page 18 of your spec file)
[image]

6. After that let's check that counter is now enabled by `hf mfu info` 
[image]

7. Now let's try read out counter by 
`hf 14a raw -sc 3900` 
`hf 14a raw -sc 3901` 
`hf 14a raw -sc 3902` 
The result will be, where 0x00 and 0x01 counters will give NAK as before, but 0x02 will give us a valid counter value
[image]

8. Now if you try read any page, for example `hf mfu rdbl -b 0` and then read counter again `hf 14a raw -sc 3902`, you will see that value has changed
[image]

RebornedBrain, 2024-09-11:
### Flipper ASCII mirror emulation
Flipper already can emulate ASCII mirror feature in all 3 modes (UID, counter, UID+counter)

Here is a short instruction, how to make it work. **Important!** To make mirror modes with counter work, you need to enable counter first (see my previous comments), otherwise it will not take effect.

1. Read clean NTAG213 and start emulation.
2. Try to dump card `hf mfu dump --ns`, you will see some data written on your card
[image]

3. Enable counter by previous instructions.
4. Read config page 41 by `hf mfu rdbl -b 41`
[image]

5. Enable mirror mode by `hf mfu wrbl -b 41 -d 840004FF` where 0x84 means mirror mode = counter and mirror byte 0, Mirror page = 0x04. 

6. Try to dump card data again `hf mfu dump --ns`, you will see that counter value in ASCII is now returned beginning from page 4
[image]

7. Reading counter value with `hf 14a raw -sc 3902` should give same counter result

ukscone, 2024-09-12:
ok. I don't actually have a proxmark so I can't use those commands to try/test but that isn't what i'm trying to do.

what i'd like to be able to do is read a tag that has nfc counter and the mirroring enabled and page/offset set correctly. save it and then emulate that saved tag and have the counter and mirroring values of the saved tag be updated when the flipper is used to emulate the tag.

The way I was trying/testing was using tagwriter, taginfo on my phone, a program running on a raspberry picow with an rc522 and the nfc app on the flipper.

my method of testing and seeing what works and what doesn't is:-

1. grab a fresh unused N215
2. read it using taginfo just to check it's in factory state
3. write a custom url ndef record with tag uid mirror enabled, counter mirroring enabled, nfc counter enabled and page and offset set to the N215
4. read the tag a few times using the picow & rc522 and check that everything is working as expected using my program to read the ndef record and config pages.
5. read the tag using the flipper, save it  and check everything is as expected
6. using the picow & rc522 to read the emulated tag using the flipper
7. check the contents of the emulated tag on the flipper

it's parts 6 & 7 that don't work as expected/wanted i.e. the contents of the emulated tag don't get updated unlike a real tag where the counter and mirror location get updated.

RebornedBrain, 2024-09-13:
Can you give me an example of your url ndef from step 3, so I can try it also?

RebornedBrain, 2024-09-13:
I've tried to write NDEF record to real NTAG with the help of this NtagWriter and it fails just as with Flipper before
[image]

For some reason this apk doesn't enable counter and mirror functions

ukscone, 2024-09-13:
yes i've had no problems using tagwriter to enable the nfc counter and counter mirror. confirmed using a simple python script and an rc522.
[image]
[image]
[image]
[image]
[image]

ukscone, 2024-09-13:
The problem/request is nothing to do with actually creating a tag with counter and mirroring. that i know how to do in several ways. using apps, manually,...  no problems what so ever.

the problem/request is that when using the flipper to emulate a cloned tag that the counter and mirroring are incremented in the saved tag file that the flipper used to emulate the tag. as i said in the 1st post I don't know if it's possible as does the flipper know that the tag has been read when the emulated tag is read but if it is it would be nice to have if it is possible

RebornedBrain, 2024-09-17:
You mean this value in flipper file remains unchanged?
[image]

ukscone, 2024-09-17:
yes.

i read a tag using the flipper that has all the relavent settings set and an ndef custom url record that has the counter mirrored and save it. then use the flipper to emulate the now saved tag and read it using a nfc reader several times and the reader is not seeing any changes in the ndef record or the counter (i check both things when reading the tag, both the ndef record and using the read counter command) so the flipper is not changing the saved file's data which makes sense as it's just a file but i would like it if the flipper if possible would change the saved file

RebornedBrain, 2024-09-17:
Ah, I got it, well mirror feature in flipper was implemented according to that spec which you've provided and there is paragraph 8.7 which says that mirror function works through virtual memory and doesn't indeed impact pages on tag. I mean this one: 
[image]

So currently mirror feature in flipper is completely dynamic and works as follow: during emulation on every read operation we check first of all does mirror feature enabled or not. If it is enabled and read command requests page which must be mirrored, then page data are dynamically replaced according to mirror feature config options. For example, when counter must be mirrored, then we format current counter to ASCII and send this formatted string as requested page content. 

And this is correct behaviour, because real cards behaves the same way (it least those which were used during implementation and testing of this feature). 

You can check this easily on the real NTAG:
1. Write your NDEF url to real NTAG
2. Enable counter and mirroring
3. Try to read dump of your card using TagInfo for example (you need Full Scan function). You will see that counter increases each time.
4. Disable mirror function but don't erase your NDEF.
5. Repeat step 3. You will see that now counter value in your NDEF is zero. In fact counter still increases and works (you can check it in Extra), but it is no longer been populated into page data during read command.

Also please pay attention to the extension of flipper file you check, because during emulation flipper doesn't use `tag_name.nfc` file, it uses `tag_name.shd` file. This file is a copy of the initial one and it is created when any data were changed on the card during emulation.

RebornedBrain, 2024-09-17:
> now saved tag and read it using a nfc reader several times and the reader is not seeing any changes in the ndef record or the counter

It looks strange, that your reader doesn't see changes, can you please provide me your tag file from flipper, to take a look

ukscone, 2024-09-17:
This might be a case of PEBKAC as I just updated to 1.0.1 firmware (i was on 0.9X something) and tried it again and it looked like it might have worked and when I went to download the saved tag a new file was also in the directory with it that i've never seen before.

i have the .nfc file and also one with the same name but with an shd extension.

the .nfc
`Filetype: Flipper NFC device
Version: 4
# Device type can be ISO14443-3A, ISO14443-3B, ISO14443-4A, ISO14443-4B, ISO15693-3, FeliCa, NTAG/Ultralight, Mifare Classic, Mifare Plus, Mifare DESFire, SLIX, ST25TB
Device type: NTAG/Ultralight
# UID is common for all formats
UID: 04 10 C8 6A 73 00 00
# ISO14443-3A specific data
ATQA: 00 44
SAK: 00
# NTAG/Ultralight specific data
Data format version: 2
NTAG/Ultralight type: NTAG215
Signature: 04 10 C8 54 6A 73 00 00 04 10 C8 54 6A 73 00 00 04 10 C8 54 6A 73 00 00 04 10 C8 54 6A 73 00 00
Mifare version: 00 04 04 02 01 00 11 03
Counter 0: 0
Tearing 0: 00
Counter 1: 0
Tearing 1: 00
Counter 2: 15
Tearing 2: 00
Pages total: 135
Pages read: 135
Page 0: 04 10 C8 54
Page 1: 6A 73 00 00
Page 2: 19 A3 00 00
Page 3: E1 10 3E 00
Page 4: 03 3E D1 01
Page 5: 3A 55 00 68
Page 6: 74 74 70 3A
Page 7: 2F 2F 62 75
Page 8: 72 6E 69 6E
Page 9: 67 68 6F 72
Page 10: 73 65 2E 63
Page 11: 6F 6D 2F 6E
Page 12: 66 63 2E 70
Page 13: 68 70 3F 75
Page 14: 69 64 3D 30
Page 15: 34 31 30 43
Page 16: 38 36 41 37
Page 17: 33 30 30 30
Page 18: 30 78 30 30
Page 19: 30 30 30 46
Page 20: FE 00 00 00
Page 21: 00 00 00 00
Page 22: 00 00 00 00
Page 23: 00 00 00 00
Page 24: 00 00 00 00
Page 25: 00 00 00 00
Page 26: 00 00 00 00
Page 27: 00 00 00 00
Page 28: 00 00 00 00
Page 29: 00 00 00 00
Page 30: 00 00 00 00
Page 31: 00 00 00 00
Page 32: 00 00 00 00
Page 33: 00 00 00 00
Page 34: 00 00 00 00
Page 35: 00 00 00 00
Page 36: 00 00 00 00
Page 37: 00 00 00 00
Page 38: 00 00 00 00
Page 39: 00 00 00 00
Page 40: 00 00 00 00
Page 41: 00 00 00 00
Page 42: 00 00 00 00
[... 232 lines cut ...]
Page 114: 00 00 00 00
Page 115: 00 00 00 00
Page 116: 00 00 00 00
Page 117: 00 00 00 00
Page 118: 00 00 00 00
Page 119: 00 00 00 00
Page 120: 00 00 00 00
Page 121: 00 00 00 00
Page 122: 00 00 00 00
Page 123: 00 00 00 00
Page 124: 00 00 00 00
Page 125: 00 00 00 00
Page 126: 00 00 00 00
Page 127: 00 00 00 00
Page 128: 00 00 00 00
Page 129: 00 00 00 00
Page 130: 00 00 00 BD
Page 131: F0 00 0E FF
Page 132: 10 00 00 00
Page 133: FF FF FF FF
Page 134: 00 00 00 00
Failed authentication attempts: 0
`

i'm not sure what the .shd file is but it seems to have an increased counter than the .nfc with everything else the same

RebornedBrain, 2024-09-18:
> I just updated to 1.0.1 firmware (i was on 0.9X something)

This explains what happened, if your were below this version [0.94.1](https://github.com/flipperdevices/flipperzero-firmware/releases/tag/0.94.1)  then your were using old nfc stack which is not actual anymore. 

> yes it does seem to be working after the upgrade of firmware and a new read of the tag. i assume the .shd file is the file created on read that has the updated counter. it's a little confusing that the ndef record is not updating in that file although the reader is reading the correct value (the counter value) from the ndef record.
> 
> it does seem to be a PEBKAC although in my defence the expected ndef record file change isn't occuring even if the counter is so it looks like it's not worked if you just look at the saved ndef record.

I think it's not your fault, this mirror mechanism is documented very poorly in specs and there are a lot of corner cases which are not mentioned there. For example, there is no any information that mirror counter feature will work only when counter is enabled. Also there is no information how tag behaves when mirror page is configured at the bottom of user memory area. And things become even worse when it turns out that different tags from different manufacturers can behave differently in those cases. Anyway I was glad to help you.

closed 2024-09-18 by ukscone