flipperdevices/flipperzero-firmware #3867: Flipper misidentifies Visonic_Powecode alarm signal as Dickhet_MAHS.

Hamika20, 2024-09-02:
### Describe the bug.

I think I've found a bug with this new protocol called Dickert_MAHS. I receive this signal multiple times a day, even though I live in a rural area far away from other garages. I receive the signal about 30 times a day. Even at midnight. When I use RTL433, the signal appears as a Visonic-Powercode sensor. I'm not sure if the Flipper is misidentifying the sensor's signal or if something else is happening.

[image]
[image]
[image]

### Reproduction

I copied the signal from the remote and tried replaying it, but RTL433 didn't recognize it at all.
I think I will need to further investigate and try to find a similar motion detector and trigger it manually.

### Target

Devs

Skorpionm, 2024-10-15:
where are the signal records in Read raw format?

Hamika20, 2025-02-07:
Well I would but I cant seem to find the source of the signal, I live in a remote area and I only have 1 neighbor who has no remote operated garage door. Its been like 5 months and I still couldn't find out the source of the signal.

Hamika20, 2025-03-29:
Aye up, I'm back after a long break, and this time I've got a RAW signal reading from the motion sensor. Got the files , plus some screenshots from RTL433 an’ all. So, I’ve linked this Dropbox link to the .sub file since I can’t upload it here.
https://www.dropbox.com/scl/fi/txqdz4gquecnwc7rhxl2v/motion_sensor.sub?rlkey=bmsrbt43k083kc7im0luzz8zv&st=jslfomiu&dl=0

[image]

Skorpionm, 2025-03-30:
no need to edit the file, need a recording of the broadcast with all the "noise" of sending repetitions and other things. and not just a "clean" signal

Hamika20, 2025-03-30:
Alright here’s the 2 files you asked for. It ain't just the clean signal, it’s the full recording of everything. One’s from the signal I recorded with the Flipper, and the other one’s from my SDR, both in my Dropbox: 
https://www.dropbox.com/scl/fi/9s8gf8ryzksev3lcfnigq/Full.sub?rlkey=oz713zm3z1ytz2utrb2mwc7zn&st=ya2pfjfg&dl=0
https://www.dropbox.com/scl/fi/niy4xkh6hpm1dnj1ww5hm/SDRSharp_20250330_153609Z_433920000Hz_AF.wav?rlkey=4aa0llmq9811mzvtouxdnm37f&st=pji486ex&dl=0
These are the files: first is the Flipper's raw read, and the second link is for the SDR reading.