flipperdevices/flipperzero-firmware #3882: GPIO USB UART Bridge disconnect's on confirm dialog

Tomi2965, 2024-09-07:
### Describe the bug.

Today I found a small bug in GPIO USB UART Bridge. 

When bridge is active, and then you press back, exit confirm dlg appears. This dialog asks left for exit right for stay. But, at this point, is bridge disconnected from USB. I think,  this is a bug. 

Many thanks for fix!

### Reproduction

For best preview, use Putty.

1. Connect to usb
2. Start Putty, select com port, open connection
3. When connection is open, press back on FZ. Exit dlg appears, and Putty will show error, that device is disconnected.

Tomi2965, 2024-09-11:
@portasynthinca3 thanks for fix! :)  I can confirm - this fix works. May I kindly ask you for small additional change? On confirm dialog - on press back key - same function as right button (stay). Thanks a lot.

closed 2024-09-12 by skotopes

portasynthinca3, 2024-09-13:
> @portasynthinca3 thanks for fix! :) I can confirm - this fix works. May I kindly ask you for small additional change? On confirm dialog - on press back key - same function as right button (stay). Thanks a lot.

Sorry, looks like I missed your request before the PR got merged

Tomi2965, 2024-09-17:
@portasynthinca3 as I see now, the PR was removed. Any chance to make a fix? Thanks a lot.

PR #3892 (2024-09-12): Fix USB-UART bridge exit screen stopping the bridge prematurely
# What's new
  - Fixed a bug where the "Exit USB-UART?" confirmation screen would stop the bridge prematurely, without waiting for the user's input. Fix #3882
  - Refactored related code

# Verification 
  - Open the USB-UART bridge
  - Configure bridge to use channel 1
  - Press the back key
  - Verify that the "Exit USB-UART" screen shows up but that the virtual COM port remains attached
  - Verify that selecting "Stay" returns to the main screen
  - Press the back key again
  - Verify that selecting "Exit" returns to the main G [... cut]