flipperdevices/flipperzero-firmware #3898: Universal Remote Feature - Enhancement Requist

HeIp, 2024-09-14:
### Describe the enhancement you're suggesting.

Currently flipper zero(Universal Remote Feature) sends a bunch of standardized codes for power on, power off, volume up/down etc....
I think for those who have lost their remotes, it would be easier to save and replay previous signals instead of having to redo the whole process to be able to replay the signals that work with their device.

Tomi2965, 2024-09-15:
Good way will be to add function for easy creating own remotes from universal codes. Just start sending in slower tempo, press button for stop the process when device receive a correct code. Then for sure allow to replay last code (or jump back), after finding. Next step is to store code into own remote (including own label).

Greendogo, 2024-09-19:
Most TVs are much more complicated than channel/volume these days and the current TV support on flipper is not helpful for anything modern. Most smart apps are based entirely off a new paradigm of standard buttons.

I would like additional buttons for these newer common menu buttons:
select, up, down, left, right, menu, back

meirm, 2024-12-15:
These are the codes for Samsung TV smart buttons:
Filetype: IR signals file
Version: 1
# 
name: Ok_samsung
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 312 293 644 1250 333 649 335 976 341 2508 360 648 390 1528 387 2179 361 2149 361 625 362 1875 367 2126 362 2451 388 85452 267 986 239 487 238 770 209
# 
name: Home
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1285 1185 500 511 500 812 406 2445 497 513 500 1414 454 2089 500 2612 500 813 525 1716 497 839 500 1713 498 86247 382 873 380 320 380 600 380
# 
name: Up
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1285 1185 500 510 500 809 445 2416 502 510 500 1413 447 2087 500 1738 474 510 501 1712 440 2685 500 2339 472 85044 384 871 380 320 380 600 380
# 
name: Down
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1285 1185 500 514 500 814 405 2448 493 513 500 1416 526 2015 500 1713 500 815 496 1714 504 2615 497 1715 497 85342 382 874 380 320 380 600 380
# 
name: Right
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1285 1185 500 512 500 814 404 2443 499 512 500 1414 453 2089 500 1710 500 1114 497 1714 511 2612 500 1112 500 85646 381 873 380 320 380 600 380
# 
name: Left
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1285 1185 500 541 471 813 407 2444 496 514 498 1418 525 2017 496 1715 497 2018 317 1952 526 2317 496 1717 494 84448 381 875 380 320 380 600 380