flipperdevices/flipperzero-firmware #3900: [Request] - Require PIN on cold boot

Devnol, 2024-09-14:
### Description of the feature you're suggesting.

I'm new to the Flipper and I've been loving it so far. I've set a PIN and it locks after I leave it idle for 30 seconds but what I find confusing is why the flipper does not ask for a PIN when cold booting from a poweroff.

From what I gather, you're not really supposed to fully turn off the device since the battery lasts for so long, but I really don't see a reason to have it running if I'm not using it for a whole day or more. After I power it on however, it doesn't ask for my PIN which I find to be a bit of an oversight if you want other people to not be able to use the device without your permission.

I believe this is a simple enough feature and while not urgent to implement, at least having the option to ask for PIN on poweron should be present.

skotopes, 2024-09-16:
@Devnol this is a bug, it is expected to ask for pin on cold

Devnol, 2024-09-16:
Interesting. What's weird is it still appears to ask for a PIN when soft reloading (left + back), but only as long as the device was already unlocked from beforehand. If you soft reload while the screen is locked it will continue to be locked after. Maybe the state of locking is persistent both in hard and soft boot? Since you can only power off the device in an unlocked state.

Devnol, 2024-09-16:
Also do verify you can repro this please. Just set a pin, power off the device and see if it asks for it.

portasynthinca3, 2024-09-16:
I was able to reproduce the issue. The device remembers whether it is locked or not using a bit stored in the RTC, which stores data for as long as the battery isn't completely dead, irregardless of whether the firmware is running or not. If it detects (using the aforementioned bit) that it was locked prior to being switched off, it will lock itself upon startup.

Here's a table summing up my experiments:
| Restart type         | Lock state prior to restart | Lock state on boot |
|----------------------|-----------------------------|--------------------|
| Soft (Left+Back)     | Locked                      | Locked             |
| Hard (holding Back)  | Locked                      | Locked             |
| Soft (Left+Back)     | Unlocked                    | Unlocked           |
| Hard (holding Back)  | Unlocked                    | Unlocked           |
| Settings > Power OFF | Unlocked                    | Unlocked           |
| Settings > Reboot    | Unlocked                    | Unlocked           |

Since it is impossible to go into the Settings menu while the device is locked, the "locked" bit is always clear upon startup after a shutdown via the Settings menu. One way to fix this would be to force-lock the device upon startup if a PIN is set up, irregardless of the value in the "locked" bit. This would also fix one other unusual scenario: complete power loss (e.g. after the device has been shitting on a shelf for a few months or worse, a bad actor disassembling the device, unplugging and then replugging the battery).

I will implement this shortly.

Devnol, 2024-09-16:
Yep, this exactly what I was thinking of, both in terms of what might be going on and how to solve it. device should re-lock on boot regardless of previous state.

closed 2024-10-16 by skotopes