flipperdevices/flipperzero-firmware #3903: Firmware update fails

kesha, 2024-09-16:
### Describe the bug.

I'm trying to update the firmware from 0.62.1 to the latest 1.0.1 version. At first with qFlipper 1.3.3 on macOS 14.6.1, M1 Pro. It seems to go without any issues, but afterwards it is 0.62.1 again.
I also tried the mobile app (iOS) to the latest version and to the custom file (version 0.63.3, https://update.flipperzero.one/builds/firmware/0.63.3/flipper-z-f7-update-0.63.3.tgz). It didn't work either. 

### Reproduction

1. Do firmware update (by qFlipper or mobile app)
2. Update fails (no errors on qFlipper, mobile app gives update error message after bluetooth reconnect)

skotopes, 2024-09-16:
@kesha 0.62.1 is quite old firmware. Is it one of the kickstarted devices?

Press and hold left + back for couple seconds, then release back, device will enter DFU mode, then use qFlipper to update device.

kesha, 2024-09-16:
Exactly, it is a kickstarted device with the last update from July 2022.
Thanks for your help, this trick worked fine.

skotopes, 2024-09-16:
Awesome, glad you update to latest version, there is a lot of stuff we shipped since 0.62 ;-)