flipperdevices/flipperzero-firmware #3907: NFC Read does not identify MF Ultralight card - OFW 1.01

kwjw3d, 2024-09-17:
### Describe the bug.

When reading an Ultralight NFC card, it only identifies the card as ISO14443-3A (Unknown) when previously it would identify this type of card correctly.

### Reproduction

Read NFC card, see Unknown instead of Ultralight

doomwastaken, 2024-10-04:
Greetings! It appears that our ultralight test cards seem to be working just fine. 
Were you able to read this card on previous version of firmware? Additionally it would be fantastic if you could tell what card are you using

kwjw3d, 2024-10-04:
Flipper read specific using 14443-3A results in unknown when NFC tools on android idetifies as ultralight
[image]
[image]