flipperdevices/flipperzero-firmware #3911: NFC Emulation Not Working Since 0.94.1

S4RP1NG0, 2024-09-19:
### Describe the bug.

Hello. NFC emulation has not been working since version 0.94.1. I updated to the latest versions, formatted it many times, tried software such as unleashed, Xtreme, RogueMaster but the result did not change. My NFC card that I use for the apartment only works in version 0.94.1.

I reintroduce the card, it registers the card without giving any errors but it does not emulate. Does anyone have any ideas?

### Reproduction

NFC Emulation not working since 0.94.1

S4RP1NG0, 2024-09-19:
Card type MIFARE Classic 1K

YR17, 2024-09-21:
Have the same issue on 101 release. Looks like communication interrupts(MfClassicListener FieldOff event received) after reading block 3(trailer block of 1st sector). The reader is accept card at 93.0.

doomwastaken, 2024-09-22:
Greetings. We would need a little more details on reader / card that you are using. Or proxmark sniff on entire communication between the devices if you have one on hand

YR17, 2024-09-22:
I could not identify reader and
unfortunately I don't have proxmark, however I made custom build based on 101 and add some logs to  inside nfc library to get part of communication. Attached the card itself. And log.
[read_log.txt](https://github.com/user-attachments/files/17088968/read_log.txt)
[card.zip](https://github.com/user-attachments/files/17088961/card.zip)

YR17, 2024-09-22:
Also I tried to log communication between flipper and card. Log attached.
[card.log](https://github.com/user-attachments/files/17088980/card.log)
A lot of messages like
`24783 [D][Nfc] FWT Timeout
24785 [T][Nfc] Failed TRX state machine`
so I suppose it could be hal related issue.

YR17, 2024-09-22:
These are my changes I gather logs at: https://github.com/YR17/flipperzero-firmware/commit/c2993ab0127430da260576077dfd520fe2f45945