flipperdevices/flipperzero-firmware #3919: C# Serial terminal not registering as computer

gameking2011, 2024-09-25:
### Describe the bug.

So, I made a serial terminal in C# since my current terminal is... well... not great. i tried it and I think the flipper will not or cannot receive the signal. i know it is sending something because when i use the usb-uart bridge i get a signal back

### Reproduction

1. download [my app](https://drive.google.com/drive/folders/1eY2xSr0XEbMKSH0D8EZhM4B8vpR8aNio?usp=drive_link)
2. open it
3. configure for flipper

### Target

Firmware

skotopes, 2024-09-25:
No google drive links please. Show us your source code on github.

gameking2011, 2024-09-26:
[here](https://github.com/gameking2011/Serial-Terminal.git)

gameking2011, 2024-09-26:
I made it in visual studio 2022

skotopes, 2024-09-26:
Letter A in UART means asynchronous, see dotnet examples on how to read data from serial port. You do need reader thread.

closed 2024-09-26 by skotopes

gameking2011, 2024-09-28:
How would I modify my code to receive a multiline string from the serial port?

skotopes, 2024-09-28:
@gameking2011 you should ask in some C# community about that.