flipperdevices/flipperzero-firmware #3920: [Sub-GHz] Request support for gate remote: 318Mhz, AM, static code

frifox, 2024-09-27:
### Description of the feature you're suggesting.

Mighy Mule gate opener remote. Sold at Home Depot.

1. Freqency: 318 Mhz
2. Modulation: Amplitude-shift keying
3. Static code
4. Raw signals:
- button 1: https://flpr.app/sf/#path=subghz/RAW-20240926-211417.sub&id=KX3dde&key=8t7sZN8oPAztPyEOdI3wRw
- button 2: https://flpr.app/sf/#path=subghz/RAW-20240926-211433.sub&id=fi67b4&key=6MeFcBCQU4B1z0wVn8IIJg
- button 3: https://flpr.app/sf/#path=subghz/RAW-20240926-211359.sub&id=L0BtCl&key=BE-liZict6nfI-kQkk3C9Q

FCC ID: 16H-MMT103

[image]

[image]

[image]

frifox, 2024-10-04:
Updated first post with new dumps, made with Universal Radio Hacker.

Seems each raw bit can be anywhere between 750us - 950us.

Then, if I assume manchester IEEE802 encoding, signals seem to consist of:
- static prefix (12 bits)
- device ID (18 bits)
- static string (10 bits)
- button ID (2 bits)

In short:
```
car button        = 111111100000 000110000101111110 0001000000 10

remote 1 button 1 = 111111100000 101011111110111011 0001000000 01
remote 1 button 2 = 111111100000 101011111110111011 0001000000 10
remote 1 button 3 = 111111100000 101011111110111011 0001000000 11

remote 2 button 1 = 111111100000 100010010000010010 0001000000 01
remote 2 button 2 = 111111100000 100010010000010010 0001000000 10
remote 2 button 3 = 111111100000 100010010000010010 0001000000 11

remote 3 button 1 = 111111100000 010001100010101001 0001000000 01
remote 3 button 2 = 111111100000 010001100010101001 0001000000 10
remote 3 button 3 = 111111100000 010001100010101001 0001000000 11
```

Skorpionm, 2024-10-15:
You decrypted almost everything yourself. What's wrong with this protocol's reception in binraw?

frifox, 2024-10-15:
Protocol works well via Read Raw. Main goal here is not to replay, but have SubGhz detect/parse the protocol and display DeviceID/ButtonID as plain-text.

For me, there are 2 reasons for that:
1. Help with switching car's remote to proper protocol before going to the gate opener to pair it.
2. I think it just too cool when you can simply "read" the remote without parsing raw dumps.

For #1, most of my cars come with remotes that support multiple protocols.
You have to press+hold each button to switch to next supported protocol and hope it matches your gate opener.
If I can use flipperzero/subghz to confirm car remote is switched to proper protocol and is transmitting properly,
that would save a ton of time having to constantly run between the car and the gate opener after every protocol switch, for each button, hoping you've matched up.

skotopes, 2025-01-18:
If someone would like to implement this protocol we'll accept it. At the same time I'll close this issue.

closed 2025-01-18 by skotopes