flipperdevices/flipperzero-firmware #3925: 0.104.0-rc branch is missing the number-input module

MassivDash, 2024-09-29:
### Describe the bug.

When trying to build my flipper application for 0.104-rc I get an error about missing number-input module. 

```
/git/flipper-tracker/app/free/../app.h:3:10: fatal error: gui/modules/number_input.h: No such file or directory
    3 | #include <gui/modules/number_input.h>
```

is the number-input being deprecated ? is there an alternative or this is an bug and the module should be there ? 

### Reproduction

- set ubft for 0.104-rc branch 
- try and build an application that uses number_input module
- Observe ubft throw error about missing module

### Target

0.104-rc

MassivDash, 2024-09-29:
The issue was raised by @furippa on my flipper app https://github.com/MassivDash/flipper-tracker/issues/1

hedger, 2024-09-29:
That module did not exist until 0.105 firmware release: https://github.com/flipperdevices/flipperzero-firmware/commit/741329a74351ffa9ac0e12f03aec1d42845f741e . 

Sorry for the automated issue, the bot that is responsible for that is not perfect and does not account for certain edge-cases.

closed 2024-09-29 by hedger