flipperdevices/flipperzero-firmware #3940: Unable to use my Chamberlain garage door with latest FW

myTerminal, 2024-10-11:
### Describe the bug.

On earlier FW versions, I was able to clone my garage remote and use the `.sub` file to operate the garage door. With the latest FW version, neither does the `.sub` file open/close the door nor am I able to intercept signals from other physical remotes I have in order to clone them. I do not even see the signals on the Frequency Analyzer, even though there are other signals that pop up in the log.

### Reproduction

1. Hit the "Center" key
2. Navigate to Sub-GHz
3. Hit the "Down" key twice to select "Saved"
4. Navigate to "Garage_Door.sub" file and hit the "Center" key
5. Select "Emulate"
6. Select "Send"

myTerminal, 2024-10-12:
[garage-door.sub.txt](https://github.com/user-attachments/files/17348834/garage-door.sub.txt)

Skorpionm, 2024-10-15:
what firmware are you using? install the official one and try to receive the signal again

myTerminal, 2024-10-15:
I am using the official FW, as mentioned above.

FW Version: 1.0.1 [10-09-2024]

Skorpionm, 2024-10-16:
Why then is the file not saved on the official firmware?

myTerminal, 2024-10-16:
> Why then was the file not created on the official firmware?

I did remember to have recorded the file using an earlier build of official FW back during my first week with my Flipper Zero, some time during the last week of December 2023. Do the contents of the file say otherwise?

Skorpionm, 2024-10-16:
yes the contents of the file indicate that you created it not on the official firmware, the original does not allow saving dynamic codes at all. When was the last time you used this file? and how many times during this time did you use the original remote? read the theory of how it works. And the file itself works
[image]

myTerminal, 2024-10-16:
It seems then that I may have recorded it during my experiments with Xtreme FW, though it also stopped working on their successor, Momentum. Would you suggest raising an issue with them? Sorry for the confusion.

myTerminal, 2024-10-16:
I'm closing this issue against [one I have created with Momentum](https://github.com/Next-Flip/Momentum-Firmware/issues/271). Thanks for your help!