flipperdevices/flipperzero-firmware #3990: BadUSB disconnecting from PC after script is run

jetrp1, 2024-11-06:
### Describe the bug.

I use the Flipper Zero to run some badUSB scripts for some game automation (single player only). I noticed this bug after updating to the current Release candidate, rc 1.1.2. When my primary script completes, often I want to run a second script which I will start from the flipper app on my phone. After the script completes, I exit the current scipt and move to the second script where I then get a prompt to connect to a computer, but a computer is already connected. The device reconnects once I reseat the usb connection.

One thing to note is that i seem to only notice this happening when it has been an extended amount of time between when the hast script completes and the new script is started. often times it is as long as overnight but can be as short as 15 to 30 mins, notably my machine is setup so as it will not go into sleep mode unless I specifcally tell it to. 

### Reproduction

1. run bad usb script
2. wait extended time frame
3. hit back button
4. choose new script
5. get connect to pc prompt

### Target

Lenovo Yoga 15-ikb running Ubuntu 24.04.1 LTS

doomwastaken, 2024-11-06:
Greetings, going to build a small test bench for this on different OS. Did it work on 1.0.1 release by any chance? 
I will get back with/for details, once I have more info available from my side.

jetrp1, 2024-11-06:
I didn't notice it, but I only had my flipper on 1.0.1 for a short time before switching to the rc channel. I can try it today though.

jetrp1, 2024-11-06:
I sweapped down to 1.0.1 and was not able to re-create the issue

doomwastaken, 2024-11-06:
In our side no luck to repeat it on rc, at least on clean Ubuntu 24.10 build. Slowly starting to think it might be some OS power management thing, will continue to check tomorrow

doomwastaken, 2024-11-07:
Extremely odd, we are not able to reproduce it, neither on laptops, nor desktops. This time tried 24.04 and updated to 24.04 1, still no luck. Will try to consult our team, maybe they have any ideas what else to we can do to verify it
Windows / MacOS works fine.

doomwastaken, 2024-11-08:
@jetrp1 Do you have an  STLink or Wifi devboard to collect logs? There is not much else that we can do on our end to verify it

jetrp1, 2024-11-11:
It might be a power issue, over the years I changed plenty of power settings and other things on this machine. The Issue wasnt very consistent, but it had happened several times. If we want to get the logs, I have a WifiDev Board, but I've never used it to get logs. I can learn though.

doomwastaken, 2024-11-12:
https://developer.flipper.net/flipperzero/doxygen/dev_board_reading_logs.html
This will help, if needed go through the section regarding dev board setup

jetrp1, 2024-11-14:
I have tried several times in the last few days but I have been unable to recreate the issue on the current RC 1.1.2 nor on the release 1.1.2 as well.

closed 2024-11-14 by jetrp1