flipperdevices/flipperzero-firmware #3997: Problem writing Electra RFID tags (but not others)

eblis, 2024-11-12:
### Describe the bug.

I have issues when trying to write an Electra tag to a T5577 tag.
Flipper is stuck in trying to write tag. If I try to read back the tag the hex value is correctly written but the epilogue is stuck at 00000...

I added some comments to #1500 but there was no answer and that issue is not specific to Electra so I thought I should open a new one.

Trying to debug the issue a bit I see that Flipper is trying to send the epilogue to the T5577 tag, but when it reads back the epilogue is all 0s for some reason.

### Reproduction

1. Get hold of an Electra tag  ID, either generate one or read an existing one (I read an existing one).
2. Try to write to an unprotected (no password) T5577 tag.
3. Flipper is stuck in "trying to write" mode and doesn't finish successfully.

### Anything else?

I've tried to debug the writing process but couldn't get it to work so I resorted to _print debugging_ it to try and find out some things.

After playing around with the logs I found out the following:
I added some logs to the encoder_start function:
```c++
bool protocol_electra_encoder_start(ProtocolElectra* proto) {
    // header
    proto->encoded_base_data = 0b111111111;

    // data
    for(uint8_t i = 0; i < ELECTRA_DECODED_BASE_DATA_SIZE; i++) {
        em_write_nibble(false, proto->data[i], &proto->encoded_base_data);
        em_write_nibble(true, proto->data[i], &proto->encoded_base_data);
    }

    for(int x = 0; x < 8; x++) {
        FURI_LOG_I(TAG, "Protocol data[%d] = %x", x, proto->data[x]);
    }

    // column parity and stop bit
    uint8_t parity_sum;

    for(uint8_t c = 0; c < EM_COLUMN_COUNT; c++) {
        parity_sum = 0;
        for(uint8_t i = 1; i <= EM_ROW_COUNT; i++) {
            uint8_t parity_bit = (proto->encoded_base_data >> (i * EM_BITS_PER_ROW_COUNT - 1)) & 1;
            parity_sum += parity_bit;
        }
        proto->encoded_base_data = (proto->encoded_base_data << 1) | ((parity_sum % 2) & 1);
    }

    // stop bit
    proto->encoded_base_data = (proto->encoded_base_data << 1) | 0;

    proto->encoded_data_index = 0;
    proto->encoded_polarity = true;

    // epilogue
    proto->encoded_epilogue = (proto->data[ELECTRA_DECODED_DATA_EPILOGUE_START_POS]);
    proto->encoded_epilogue <<= 8;
    proto->encoded_epilogue |= (proto->data[ELECTRA_DECODED_DATA_EPILOGUE_START_POS + 1]);

    FURI_LOG_I(TAG, "Initial encoded epilogue %llx", proto->encoded_epilogue);

    //fill bytes 2-7 by epilogue filler
    for(uint8_t i = 2; i < ELECTRA_ENCODED_EPILOGUE_SIZE; i++) {
        proto->encoded_epilogue <<= 8;
        proto->encoded_epilogue |= proto->data[ELECTRA_DECODED_DATA_EPILOGUE_START_POS + 2];
    }

    FURI_LOG_I(TAG, "Base data %llx", proto->encoded_base_data);
    FURI_LOG_I(TAG, "Encoded epilogue %llx", proto->encoded_epilogue);

[... 144 lines cut ...]
34671 [D][ELECTRA] Epilogue[2] = 00000000
34673 [D][ELECTRA] Epilogue[3] = 00000000
34676 [D][ELECTRA] Epilogue[4] = 00000000
34678 [D][ELECTRA] Epilogue[5] = 00000000
34681 [D][ELECTRA] Epilogue[6] = 00000000
34683 [D][ELECTRA] Epilogue[7] = 00000000
34685 [T][ELECTRA] Decoded data size 8
34687 [I][ELECTRA] Initial data[0] = 00000003
34690 [I][ELECTRA] Initial data[1] = 000000e8
34692 [I][ELECTRA] Initial data[2] = 00000089
34695 [I][ELECTRA] Initial data[3] = 00000048
34697 [I][ELECTRA] Initial data[4] = 000000fc
34700 [I][ELECTRA] Initial data[5] = 00000000
34703 [I][ELECTRA] Initial data[6] = 00000000
34705 [I][ELECTRA] Initial data[7] = 00000000
34708 [D][ELECTRA] ELECTRA_ENCODED_EPILOGUE_SIZE 8 VS encoded_epilogue_size 8
34711 [W][ELECTRA] Final Data[0] = 00000003
34714 [W][ELECTRA] Final Data[1] = 000000e8
34716 [W][ELECTRA] Final Data[2] = 00000089
34719 [W][ELECTRA] Final Data[3] = 00000048
34721 [W][ELECTRA] Final Data[4] = 000000fc
34724 [W][ELECTRA] Final Data[5] = 00000000
34726 [W][ELECTRA] Final Data[6] = 00000000
34729 [W][ELECTRA] Final Data[7] = 00000000
```

eblis, 2024-11-12:
From what I can tell @Leptopt1los added the code for Electra protocol, maybe you have some ideas.
I don't have a WiFi board which, from what I understand, makes debugging the code a lot harder so I resorted to print debugging various parts of the code.

[These are the T5577 tags I bought](https://www.aliexpress.com/item/1005006353951182.html?spm=a2g0o.order_list.order_list_main.5.2aea1802P54pDm). I understand that someone recommended some other tags [from eMag](https://www.emag.ro/set-10-bucati-tag-de-proximitate-rfid-e-locks-125-khz-chip-t5577-rewritable-galben-t5577elck/pd/D5RK86MBM/) in another issue but @itrack - who seems to have the same issue as me - tried those tags as well without success either.

itrack, 2024-11-12:
@eblis I tested these tags from eMAG, same problem, they don't work

skotopes, 2024-12-02:
@Astrrra @doomwastaken can you take a look?

Astrrra, 2024-12-05:
@eblis @itrack hello! It seems that this behavior is caused by non-compliant T5577 tags. I have one such tag in my possession on which I can reproduce the issue, but I can't reproduce it on any other T5577 tags I own. On all the other tags the whole epilogue gets written completely and as expected.

Do you happen to have a proxmark3 to diagnose your T5577 tags? If not - I'm afraid that the only solution to this is to get T5577 tags from a reputable supplier (I can confirm that everything works as expected on T5577 tags from lab401.com) or to check them before buying if possible.

eblis, 2024-12-05:
Unfortunately I do not have Proxmark3.
I can change the code in flipper and test various things if that helps.

How do I test that the T5577 tags are compliant ? What am I looking for ? Or just that it can write an Electra tag, including the epilogue?

Astrrra, 2024-12-05:
Yes, just writing the Electra data should be enough in your case

eblis, 2024-12-05:
Ok, I have some great news, I think :)

Since you commented that it may be a T5577 issue I tried again, and of course the first 3 tags I tried I couldn't write, they were the same tags I had tried before.
But the 4th tag that I tried worked !! It's from the same order as the first 3 tags I had tried, but for some reason this one works.
I tried one more, also worked. Tried 4 more, didn't work.

I have a mixed color set and looks like I can write to the gray and black ones, but not any other color.

eblis, 2024-12-07:
I would say that everything works then, it's a matter of using the correct T5577s'.

closed 2024-12-07 by eblis