flipperdevices/flipperzero-firmware #4001: NiceFlo cant decoded in Sub-GHZ

Penelopen, 2024-11-13:
### Describe the bug.

Nice Flo 12/20bit or Nice FloR 56bit are not decoded in flipper. It not know about these systems, but I can create this remotes manually..

Even if I manually create these remotes in flipper, that my another flipper cant see that.

Flipper can create Nice Flo/FloR but can't decode it.

### Reproduction

Open Sub-GHZ Read

### Target

Sub-GHZ

Penelopen, 2025-01-16:
1. I just tried to read Nice Flor-S and Nice Flo remote. Flipper cant decode it.
2. I manually created an Nice Flor-S 433Mhz Remote on Flipper (A) and then try to read it on Flipper (B) and it cant read this. Even if I read RAW, then it cant decode it.

Maybe it happens with other Nice systems.. But Nice Smilo is OK.

skotopes, 2025-01-16:
We need more details ) which specific remote, which system?

Penelopen, 2025-01-16:
What you mean about system?)) Nice Flor-S... Just create that in your flipper and then try to decode))

I not have that remote. One year ago I copyed and saved it in CodeGrabber and now I repeat that signal to Flipper, but Flipper cant decode it. Flipper successfully decoded all of 30 remotes from CodeGrabber, but cant decode Nice Flor-S.

Then I try to manually create this Remote in Flipper and it also cant decode own Remote. Try it yourself))

skotopes, 2025-01-16:
@Skorpionm FYI

Skorpionm, 2025-01-17:
what kind of firmware do you have? I can't be responsible for the fact that other people's firmware doesn't work. Nice FloR S, this is a rolling code system and the creation of keys for this system is not supported.

[image]

[image]

Penelopen, 2025-01-17:
I have RogueMaster installed. They also says, that they cant modify fundamental features and send me to you.

"we don't have support to expand upon the current implementation and we did not create it. So therefore, this is not the place for this ticket."

Tomorrow I get this NiceFlor signal with that problem and post it .SUB here. And you can see, that it cant decoding

Skorpionm, 2025-01-17:
I showed that Nice Flo works on the official firmware

Penelopen, 2025-01-18:
Here is video about that bug and .sub. I repeat, that I also try it with unleashed...

[RAW-20250118-231614_NiceFlor.zip](https://github.com/user-attachments/files/18466005/RAW-20250118-231614_NiceFlor.zip)

https://github.com/user-attachments/assets/714f6f7c-57f1-47d8-bb4d-26eb50bb16b5

Penelopen, 2025-01-18:
And here is some more problem's RAW. Now it is **NiceFlo**

Both of this signals successfully works on black tranciever

[RAW-20250118-231804_NiceFlo.zip](https://github.com/user-attachments/files/18466010/RAW-20250118-231804_NiceFlo.zip)

skotopes, 2025-01-18:
@Penelopen we don't provide support for 3rd party firmwares

closed 2025-01-18 by skotopes

Penelopen, 2025-01-18:
You not hear me. **I TRY IT IN YOUR ORIGINAL FIRMWARE**! THERE IS NO RESULT! TRY IT YOURSELF

Для тугоухих: Я проделывал это на вашей оригинальной прошивке! Всё то же самое! Вы, как принципиальный ребёнок не хотите вникать, потому, что у меня установлена другая прошивка! Но на оригинальной тоже самое происходит!