flipperdevices/flipperzero-firmware #4002: RFE: Cannot copy NFC-V tag

marcmerlin, 2024-11-13:
### Describe the enhancement you're suggesting.

### Describe the bug.

I'm using the latest available firmware as of today. I have an NFC-V tag on a water filter that I'd like to duplicate to a blank tag, but copying is not offered for this type of tag, only emulation and that emulation didn't seem to work (my water filter didn't see the emulated tag at all, or didn't like it).

Old versions of flipper zero recognized the tag as ISO15693-3 (unknown)
[image]

New (today's) firmware, recognizes the tag as SLIX
[image]

For comparison, the same tag seen by android
[image]

### Reproduction

Reading the tag works, emulating it seems to work, but at least in my case does not. I can't tell if the emulation is bad or if I have another issue.
But more importantly, there is no 'copy' option that I can find.

I bought these 2 kinds of tags, but currently cannot write to them:
https://www.amazon.com/dp/B0755WF6CK
and https://www.amazon.com/Ntag215-Stickers-Guaranteed-Perfectly-Compatible/dp/B07P6WZ1JX

Any idea what I'm doing wrong, or not supported yet?

### Anything else?

I may have filed this in the wrong repo originally, namely https://github.com/DarkFlippers/unleashed-firmware/issues/836
apologies for that.
I'm happy to close the other one if appropriate

skotopes, 2025-01-16:
@RebornedBrain can you take a look?

RebornedBrain, 2025-01-20:
Hi, @marcmerlin currently Flipper doesn't support write operations for iso15 tags. 
Speaking about emulation, do you have proxmark? If yes, then some sniff result logs from it could be very handy.
You can also try next steps:

- Read the tag and save it on FZ
- Download saved tag.nfc to your PC and open it with any text editor
- Change `Capabilities` field from `Default` to `AcceptAllPasswords` here is an example 
[image]

- Save your tag and upload it back to FZ
- Try to emulate