flipperdevices/flipperzero-firmware #4003: Ability to make custom menus

GonzalesGabe, 2024-11-15:
### Description of the feature you're suggesting.

A new functionality to create custom menus for different things such as games, and thr ability to customize what you want in the menu by choosing an entire folder so it includes all the contents of the folder or actually hand picking the files/apps you want, you can choose when you want it to open like on boot or if you need to enable it and have a special code to go back to normal or something. I think this is a cool splash of customization that would be really cool and convenient.

skotopes, 2025-01-16:
We'll try to implement it In the next flipper generation

unnamedd, 2025-01-16:
> the next flipper generation

😱