flipperdevices/flipperzero-firmware #4007: universal IR

ddddddddggfekzurkj, 2024-11-18:
### Description of the feature you're suggesting.

add to the universal IR remote control the ability to pause, scroll to the previous or next signal, save it and send the desired signal, like in the sub-ghz bruteforcer application

### Anything else?

добавте в универсальный ик пульт возможность приостановить, перелистнуть на прошлый или следующий сигнал, сохранить его и  отправить нужный сигнал типо как в приложении sub-ghz bruteforcer

skotopes, 2025-01-16:
@portasynthinca3 wanna try something new?

portasynthinca3, 2025-01-29:
Hey @ddddddddggfekzurkj! Please install the firmware from #4085 ("Web/App Updater" option) and check whether it's what you wanted.

Пожалуйста, установите прошивку из #4085 (пункт "Web/App Updater") и сообщите, это ли Вы просили.

ddddddddggfekzurkj, 2025-03-22:
yes @portasynthinca3 this is what I asked for, it turned out really cool, thank you

да @portasynthinca3 это то что я просил получилось очень круто спасибо

agathakazar, 2025-04-17:
This is great! 
Would it be possible to have the protocol and the hexes be displayed in this pause menu? 
Much more convenient than going to ctrl+f the tv.ir file

agathakazar, 2025-05-20:
Also, Flipper hangs when I `ir universal tv Power` from the cli, feels like it's related (does it warrant a new issue? no idea)

mirkokral, 2026-09-30:
> Also, Flipper hangs when I `ir universal tv Power` from the cli, feels like it's related (does it warrant a new issue? no idea)

Universal IR takes pretty long to transmit all codes, just wait probably unless it takes like very very long