flipperdevices/flipperzero-firmware #4008: USB-UART Bridge error

EstebanFuentealba, 2024-11-18:
### Describe the bug.

Hi,

I have a board (MALVEKE) where I’ve always flashed its [firmware](https://flipperzero.json.cl/) using the Flipper Zero’s USB-UART Bridge.

Previously, I was using firmware version 0.105.0, and everything worked perfectly.

Recently, I updated to version 1.1.2 and tried flashing the firmware to my boards, but now I’m encountering errors.

I’ve tested with several MALVEKE boards I used previously for development and troubleshooting. I even swapped the USB cable to rule out any issues with it, but I’m still frequently seeing errors like the following:

[https://www.youtube.com/watch?v=RtA6t9S1A90]([image])

[https://www.youtube.com/watch?v=FLeudK6qEv4]([image])

### Reproduction

1. Connect the module to the Flipper Zero.
2. Navigate to `GPIO` > `USB-UART Bridge`.
3. Put the board into boot mode.
4. Flash the firmware.

### Target

GPIO

kbembedded, 2024-11-19:
To add some context, additional testing, and other references, here are links to it in flipper discord:
https://discord.com/channels/740930220399525928/967843558520418384/1308199204719431731
https://discord.com/channels/740930220399525928/967843558520418384/1308520453848961094
https://discord.com/channels/740930220399525928/765282833744265246/1308166431896244264

GangsterFox, 2024-11-19:
Adding some context to the third link above.
The issue there which I am facing is the lack of data accuracy and data loss at higher baud rates using the UART to USB Bridge, similar to [this issue](https://github.com/flipperdevices/flipperzero-firmware/issues/2304) and shouldn't be focused on in this issue.

valievdmitrii, 2024-11-20:
i have this issue too. 
by steps:
1) open gpio -> uart-bridge
2) conect (for ex. openwrt board to uart)
3) open putty and open connect
4) input "dmesg" 
5) after 20-30 lines putty show message - comport is out --> fliper is die.

FW ver: 1.1.2 HW 12.F7B9C6

APF-MRoger, 2024-11-20:
Adding to this, trying to flash a ESP32 board over UART using the Flipper in USB-UART mode (so USB to PC and 13/14 to the ESP32), the flipper crashes. Sometimes right at the flashing start, sometimes at the baudrate change to higher speed for the actual bulk of the flashing.

doomwastaken, 2024-11-21:
Greetings, we will verify it and if issue is there, will add it to test cases / create a ticket for it. Will update on it later

AeneasChristodoulou, 2024-11-21:
> Adding to this, trying to flash a ESP32 board over UART using the Flipper in USB-UART mode (so USB to PC and 13/14 to the ESP32), the flipper crashes. Sometimes right at the flashing start, sometimes at the baudrate change to higher speed for the actual bulk of the flashing.

I am experiencig the exact same issues, reading values from UART (e.g. Serial output) works with my ESPs, but if you want to flash the firmware or sometimes even on Baudrate change, the Flipper will crash with differing Error messages including hard fault.

dshcherb, 2024-11-27:
Same here, very easy to trigger even on basic transfers: tested with different devices like a WaveShare's USB UART (PL2303). Sending a bunch characters via the keyboard causes a crash pretty quickly - don't even need to pass a lot of data.

Definitely started happening after the upgrade to 1.1.2.

modemrunner, 2024-12-01:
Same here, with 1.1.2 firmware. Crashed with raspberry Pi 3 with Hard Fault error, 115200 baudrate. after some amount of RX data.

doomwastaken, 2024-12-01:
We are already working on fix.

Astrrra, 2024-12-02:
@EstebanFuentealba uploaded a fix to #4025, pls check if that one works

modemrunner, 2024-12-02:
> @EstebanFuentealba uploaded a fix to #4025, pls check if that one works

For me it work very well! Used with Raspberry Pi 3, no crash detected. Thank you for your work!!!

closed 2024-12-18 by skotopes

EstebanFuentealba, 2024-12-19:
I tested again with a widely used firmware like [CameraSuite](https://github.com/jamisonderek/Flipper-Zero-Camera-Suite), but now the Flipper freezes. I think it has so many features now that there's no memory left 🥲.

https://github.com/user-attachments/assets/8dbdd36f-d42f-4df3-882b-195fed84ccd6

EstebanFuentealba, 2024-12-19:
This works [CameraSuite](https://github.com/jamisonderek/Flipper-Zero-Camera-Suite) with firmware version 1.0.1.

https://github.com/user-attachments/assets/e2334389-6e45-457b-bad6-8217f692e185

doomwastaken, 2024-12-20:
Will do some additional tests then, lets see if anything else is affected. We could probably rule out low memory possibility by taking latest tag firmware and clearing some space there

skotopes, 2024-12-23:
@EstebanFuentealba can you also show us serial logs from your device? (change log serial port in setting if you need main serial for your extension board)

EstebanFuentealba, 2024-12-23:
> can you also show us serial logs from your device? (change log serial port in setting if you need main serial for your extension board)

Are you referring to the `ufbt cli` -> `log trace` logs? Or to connecting an ST-Link for debugging?

PR #4025 (2024-12-18): Increase system stack's reserved memory size
# What's new

- Fixes #4008 by partially rolling back changes made in #3924 

# Verification 

- See #4008 

# Checklist (For Reviewer)

- [x] PR has description of feature/bug or link to Confluence/Jira task
- [x] Description contains actions to verify feature/bugfix
- [x] I've built this code, uploaded it to the device and verified feature/bugfix