flipperdevices/flipperzero-firmware #4009: [mbedtls] Add AES part of the library.

csaboka21, 2024-11-18:
### Describe the enhancement you're suggesting.

Hi,

I would like to ask to add AES implementation of the mbedtls library.
I would use it in my own app.

GMMan, 2024-12-31:
There are built-in AES functions within furi HAL (see `furi_hal_crypto.h`), see if those match your needs first. If they don't, you will need to ship your own copy of mbedtls. The built-in mbedtls functions are not exported to the application API.

bettse, 2025-01-03:
`mbedtls_des_crypt_cbc` is included in the API (via `fap_libs=["mbedtls"],`), I don't see why the AES variants couldn't be made available as well.  I would find them useful.  There is `furi_hal_crypto_decrypt`, but that appears to only support AES256, not AES128.  I'd be more than happy to be mistaken though.

GMMan, 2025-01-03:
The hardware definitely supports AES128, so maybe it's a small change to enable that in FuriHal. I think hardware AES peripheral use should be encouraged. Having to link mbedtls also bloats the app executable size.

bettse, 2025-01-04:
Sure, I don't care how, I'd just like AES128.  If they can do it in hardware, that's great, otherwise, mbedtls, and the bloat it adds to my app, would be something I'd accept.

skotopes, 2025-01-16:
@portasynthinca3 couple things here:
- update mbedtls to latest version
- update mbedtls build configuration
- optional: talk with @hedger about impact on firmware

portasynthinca3, 2025-02-13:
Implemented in #4092

closed 2025-02-13 by portasynthinca3