flipperdevices/flipperzero-firmware #4016: BadUSB F13-F24 keys support not functional

jezh42, 2024-11-25:
### Describe the bug.

When trying to use BadUsb and badusb-js on the latest version of the firmware, I am unable to properly emulate the F13-F24 keys. Support for these keys were apparently added back in [March](https://github.com/flipperdevices/flipperzero-firmware/pull/3468), and the code has not been removed since. 
When executing a BadUsb script with F1->F24 connected to a Windows or Linux machine, with a supported keyboard input program monitoring keypresses, I am unable to register any of the F13-F24 keys.
Using other software, such as AutoHotKey, I am able to register F13-F24, so the monitoring program is not the issue. Furthermore, Windows has support for F13-F24 shortcuts, such as the default enabled Win + F21 opening settings, which I am also unable to emulate through Flipper BadUsb.
I have tried using badusb-js as well, trying F13-F24 directly, HID decimal and HID hex and no success.

### Reproduction

1. Create a BadUsb script containing F13 up to F24, each on new lines, followed by "GUI F21" on the final line
2. Copy script to Flipper
3. Plug Flipper into a Windows computer via USB
4. Open a keyboard tester program that supports F13-F24
5. Execute the BadUsb script via Flipper
6. Notice the keypresses triggered, only the GUI key will be executed

skotopes, 2025-01-16:
We'll keep it in backlog

rseufert, 2026-10-16:
This looks fixed as of 1.5.0. Root cause: both HID keyboard report descriptors capped the key array at usage/logical maximum 101 (0x65), and F13..F24 are usages 0x68..0x73, so hosts discarded them even though the parser and key table were right. #4361 (commit 9148d50c, "Fix USB/BLE HID limits to support international keyboards") raised both limits to 255 in targets/f7/furi_hal/furi_hal_usb_hid.c and lib/ble_profile/extra_profiles/hid_profile.c, and that commit is in 1.5.0-rc, 1.5.1-rc and 1.5.1.

Verified on a Flipper Zero running 1.5.1 with a BadUSB script of F11, F12, F13..F24, F12 against a macOS host with a system-wide key event tap: F11, F12 and F13..F20 all arrive with the expected key codes (macOS has no virtual key codes for F21..F24, so those can't be observed that way, but they sit in the same now-allowed range). Probably safe to close.