flipperdevices/flipperzero-firmware #4019: Cannot read the SIMON VOSS RFID CHIP

Killer74-hub, 2024-11-27:
### Describe the bug.

Chip: simonsvoss rfid chip 3074 CH

### Reproduction

1. press / press and hold
[image]
the button on the remote rfid chip
2. 2. hold the Fz on back of the chip
3. Nothing…

Varbin, 2024-12-05:
According [to this forum posts](https://forum.flipper.net/t/frequency-simons-voss-fob/5976) the transponder uses 25 kHz, which isn't supported by the Flipper.

skotopes, 2025-01-16:
Yep there are limitation on our side:
- what you trying to read MUST continuously repeat payload
- it must use 125kHz carrier

Looks like your device fails both.

closed 2025-01-16 by skotopes