flipperdevices/flipperzero-firmware #4031: NFC Plugin for type 4 tags

bettse, 2024-12-11:
I was looking into writing a plugin for a type 4 tag (iso14443a-4, speaks APDUs).  The plugin's parse method receives a `NfcDevice *`, but looking through the NFC code, I can't see a way to go from that to a `Iso14443_4aPoller *` poller so I can call `iso14443_4a_poller_send_block`.  

I presume the parse is in something akin the NFC callback where one would normally do this, but it doesn't have the same event that the poller is pulled from in the other code I've written.

Is getting the poller possible, or is there something that needs to be changed in the firmware?

RebornedBrain, 2024-12-20:
Hi, `NfcDevice*` is a representation of a card with all its data, and poller creates an instance of empty card data and then when read is complete we copy data from poller to `NfcDevice` for example like this:
[image]
Then parsing with plugins uses that `nfc_device` where we have card data.

If you want to send some additional frames to card, it looks like you need some custom poller on top of `iso14443a-4` and once it finishes then you provide data from it to plugin.

Can you please describe a little bit what you want to do? Maybe I can help with some better advices.

bettse, 2024-12-20:
It sounds a lot like there is no way to send iso14443a-4 frame from within a plugin?

> Can you please describe a little bit what you want to do? Maybe I can help with some better advices.

I'd like to select an AID: `0x00, 0xa4, 0x04, 0x00, 0x0a, 0xa0, 0x00, 0x00, 0x04, 0x40, 0x00, 0x01, 0x01, 0x00, 0x01, 0x00` and then make further commands after that.

RebornedBrain, 2024-12-24:
I understand, well technically I see some ways, how poller can be provided to plugin, but all of them look very odd.

1. You can read card as  iso14443a-4, then pass data to your plugin, where you can create your custom poller and use NfcDevice data with it. But in such case in your plugin you will have to re-read card completely again, because first poller (before plugin) will be stopped and there will be no field which powered your card. So you will have to create your poller, activate card again and after that do all the job.
2. You can create a plugin with another interface, where you can provide all things you want. But in such case there is a question how to insert this new plugin in our current nfc_app logic, because currently we read card data and then iterate through plugins until we found matching one. As a workaround this can be done through our 'Extra Actions' menu, where we can add `Read as 'Your card'` button. In general this idea looks interesting to me, because it opens an ability to add new pollers without changing nfc library, but it requires more research in order to be sure this will do the thing.

From these 2 options I would choose the second one if there is no others😸, but you can also try such thing which, in my opinion will be better, rather than previous  workarounds upper:

1. Add new poller to our lib, or try to modify existing one. (I don't know how huge is the amount of logic you want to add, so propose both options).
2. Think of implementing your logic as a separate app. In such case you can build your custom poller the way you want, and be almost off the constraints you have right now. By the way, as an example you can check this [app ](https://github.com/flipperdevices/flipperzero-good-faps/pull/245/files#top) here I've done something similar with iso14443_4a poller for nfc eink displays. In this [file](https://github.com/flipperdevices/flipperzero-good-faps/pull/245/files#diff-1a19522b2d5fca8c73d8e672c7b338fab49144e1d2a3a71f289e8491ba42bf03) there are some typedefs of apdu structs, and [here](https://github.com/flipperdevices/flipperzero-good-faps/pull/245/files#diff-a206802239f91ed86f84b2c072976711ec5ead2fc91cd6b33669dc8f6b00f642) there is a poller itself, which uses them. Maybe it might help.

P.s. Idea of 'Poller as a plugin' looks interesting, I will talk to other nfc devs, about it

bettse, 2024-12-24:
Thanks for the response, yeah a lot to think about.  

One more thing on this topic: am I crazy, or is the iso14443a-4 working different in this case compared to MFC or desfire?  I looked through some of those plugins and it seemed like you could auth and read data from mifare in your plugin, but maybe I was misunderstanding?  or for desfire I saw a function about selecting an AID.  

> Think of implementing your logic as a separate app.

To give more context for what I was trying to do: I'd hoped to write a plugin for the main NFC app to detect a card type and tell the user to investigate the specific app.  There are lots of type 4 tags out there where you don't want to write all the support in a plugin, but for the best user experience, if the main NFC app could point them to the right app to look into, that'd be a win.  

I'm going to close this, given I have an answer for the time being.

closed 2024-12-24 by bettse