flipperdevices/flipperzero-firmware #4042: Add Walkie Talkie recording

Killer74-hub, 2024-12-20:
### Description of the feature you're suggesting.

Use the files from:

https://github.com/jamisonderek/flipperzero-firmware-walkietalkie/tree/jamisonderek/walkietalkie

profilelag, 2025-01-10:
Doesn't sound like something that will be added to OFW, maybe a 3rd party app?

Killer74-hub, 2025-01-10:
> Doesn't sound like something that will be added to OFW, maybe a 3rd party app?

Hm. But then someone needs to do it because I still need to learn the programming language.

skotopes, 2025-01-16:
Record RAW is kinda what you want. However we don't plan to add to SubGHZ app and I hope that someone from community will make separate app for that.

closed 2025-01-16 by skotopes

Killer74-hub, 2025-01-16:
> Record RAW is kinda what you want. However we don't plan to add to SubGHZ app and I hope that someone from community will make separate app for that.

Sadly record RAW doesnt Work.