flipperdevices/flipperzero-firmware #4044: Add ISO 3166 to ISO 11784 and ISO 11785 animal reading in the 125 kHz RFID app

Jolexer, 2024-12-23:
### Describe the enhancement you're suggesting.

When reading an animal as described in [https://docs.flipper.net/rfid/animal-microchips](https://docs.flipper.net/rfid/animal-microchips), the country code it displays is non-descript. These country codes are defined in [ISO 3166](https://docs.flipper.net/rfid/animal-microchips) (wikipedia). Please display the country number along with the english country name in the full information display. In the abreviated info screen, it may be better to show the ISO 3166-A-2 country abbreviation also defined in that wikipedia page, instead of the full english name, to better fit the UI.

Below is a are current screenshots and their proposed versions:
[image]
[image]
[image]
[image]

### Anything else?

I am really grateful that the flipper can read dog tags. Many people don't return collarless dogs because they don't want to bring them to a vet. Having given this feature to the flipper wil help many dogs find their owners.

equipter, 2024-12-23:
worth noting the first 3 digits "country code" are not always a country code. 

https://mypethq.io/a-guide-to-pet-microchips/

so if this is added it might be worth it to replace country code with manufacturer code if the 3digit code found is not a country code.

Jolexer, 2024-12-23:
> worth noting the first 3 digits "country code" are not always a country code.

That is true, but they all start with 9xx. You could either put the mfg names in the list as well, or just display "invalid" instead of a country (which should be done for invalid codes anyway).

skotopes, 2025-01-16:
I'll put it into backlog, but no promises.

closed 2025-09-25 by hedger