flipperdevices/flipperzero-firmware #4045: [NFC] Protocol st25tb write data 0 at block when not initialized

yashikada, 2024-12-24:
### Describe the bug.

I'm developing a simple app for read tag ST25TB, after I started the poller with callback function where the NfcCommand is always Continue and no events are managed, start a write at block 0 with data 0x000000.
In that device block 0x0 is OTP so is irrecoverable unless you use a sequence of commands to reset the OTP area to the value 0xFFFFFFFF.

In file lib/nfc/protocols/st25tb/st25tb_poller.c function [st25tb_poller_request_mode_handler](https://github.com/flipperdevices/flipperzero-firmware/blob/5fb9558dbb1b48dca1d48e6bffad9f7bf9d9f7a0/lib/nfc/protocols/st25tb/st25tb_poller.c#L96) should be check also the mode St25tbPollerModeWrite and don't assume it's Write if it's not Read.
In my test code the value of variable st25tb_event->data->mode_request.mode is 4, which means no mode, but everything that is not Read is automatically Write.

### Reproduction

1. develope an example NFC app which use St25Poller
2. compile
3. run
4. put a tag ST25TB under FZ
5. verify if the block 0 is written to 0x000000, you can see also in log debug

### Logs

```Text
84872 [D][Nfc] FWT Timeout
84883 [D][ST25TBPoller] Read_block(0) result: 00000000
84886 [D][ST25TBPoller] wrote 00000000 to block 0
```

skotopes, 2025-01-16:
@RebornedBrain FYI

augustozanellato, 2025-01-16:
I don't feel like this is actually a bug, the only valid operations you can request from the poller are read or write, if you don't intend to do any of those you might as well just return `NfcCommandStop` and stop it.

yashikada, 2025-01-17:
>  you might as well just return `NfcCommandStop` and stop it.

now that I have developed the first nfc yes FZ app I know it too, but at the beginning I didn't know it and I was doing tests with the firmware release. After opening the issue I saw that furi_assert is only compiled as a firmware debug, even putting furi_check would be fine.
I opened the issue to prevent what happened to me from happening to others, if it wasn't an OTP area it wasn't a problem.

yashikada, 2025-01-21:
4 is the default value, I don't request nothing, the callback function return continue and print status.
So without assert, it's wrong to think if is not read is write.
For me that code is bad.

augustozanellato, 2025-01-21:
There are only two valid values so it’s safe to assume that if it’s not one it’s the other.
The check for valid values is there, but as I said it’s only checked on debug builds.

RebornedBrain, 2025-01-28:
Hi, @yashikada, thanks for pointing this out. Best practice is to use DEBUG=1 during development process, but here maybe we will replace furi_assert to furi_check in order to prevent rokies from damaging OTP on their cards.