flipperdevices/flipperzero-firmware #4055: U2F WebAuthn Doesn't Work on MacOS 15.2

DustyArmstrong, 2025-01-01:
### Describe the bug.

Flipper's U2F no longer works on MacOS >15.2 with WebAuthn. The device no longer prompts to press the button when trying to authenticate with it as a security key, it just stays on the "Connected" screen. 

Strangely, you _can_ still set up the device as a security key, but it simply no longer prompts to authenticate once registered. 

Running latest available firmware. Have tried on multiple Macs running 15.2, Windows 11 still works fine. Not browser-specific issue, Firefox and Edge both work in W11, it's only MacOS. I assume they have created some new auth implementation as they are trying to push Passkeys hard.

### Reproduction

1. Open U2F app
2. Go through the authentication process until the "Press button" prompt should appear
3. The prompt will never appear and you cannot authenticate

DustyArmstrong, 2025-01-01:
Did a bit more digging, this is an issue with [FIDO over FIDO2](https://support.yubico.com/hc/en-us/articles/17388309240348-Safari-18-2-MacOS-iOS-iPadOS-FIDO-known-issues). 

Appreciate this isn't high (or even) on your roadmap, so I'm happy to close this unless there's likely to be any development on it, I understand there are obstacles to implementing FIDO2.

skotopes, 2025-01-16:
It does sounds like fix is coming on soon and already available in beta:

> January 10, 2025 update: In Apple's macOS 15.3 Beta 2 (24D5040f) and iOS/iPadOS 18.2 Beta 2 (22D5040d) releases, there are fixes for both issues documented in this article. Please test within your environment, [report any issues to Apple](https://developer.apple.com/bug-reporting), and subsequently inform Yubico including your Apple issue/feedback identifier.

I'll close this issues since it is not related to Flipper U2F application

closed 2025-01-16 by skotopes