flipperdevices/flipperzero-firmware #4059: ISO14443a-4 emulation

bettse, 2025-01-03:
Is it possible to send a response in a NfcProtocolIso14443_4a listener callback?  I didn't see any equivalent to `iso14443_4a_poller_send_block` and `iso14443_3a_listener_send_standard_frame` which is used by `iso14443_4a_listener_send_ats` is private (in `lib/nfc/protocols/iso14443_3a/iso14443_3a_listener_i.h`).

skotopes, 2025-01-16:
@RebornedBrain @gornekich FYI

RebornedBrain, 2025-01-21:
Hi, @bettse, yes currently it's not possible to send frames from `iso14443_4a_listener.c` and what kind of response do you need? If it is a part of iso14443_4 spec maybe it should be done in a same manner as `iso14443_4a_listener_send_ats`

bettse, 2025-01-21:
> yes currently it's not possible 

I'm glad I didn't just miss it :+1:

> what kind of response do you need?

I think it'd be called a "R-APDU".  I'm looking to do some type 4 emulation, so when the flipper gets the select application, I need to respond with either a file not found (6A 82) or the appropriate "Answer to Select".  I've got a couple ideas for stuff to emulate, but the easiest to discuss publicly would be emulating the NTAG424.

Seeing as how this would be a new feature, feel free to close this issue out if you'd like to track it through some other system or issue.

RebornedBrain, 2025-01-23:
Hi @bettse , thanks for poiniting on NTAG424, we will take a look at them and try to implement. And yes from 14443-4a emulation only ATS was implemented. Looks like it's time to extend its functionality, we will take to the backlog 👍

bettse, 2025-01-27:
For anyone else interested, you can go a step lower and use the `nfc_listener_tx`, although this requires you handle the 2 bytes before the ISO14443a-4 (APDU), which look familiar, but I don't recall the exact specification for them.

https://github.com/Serialbocks/nfc_tesla/blob/main/helpers/nfc_tkc_listener.c#L139C1-L147C10

KyoungsueKim, 2025-02-22:
Did you know that in earlier Flipper Zero firmware versions, you could use the CLI to send APDU commands and check the responses? I also tested APDU commands using the CLI in the past, but when I recently tried to use it again, I found that all NFC CLI-related functions had been removed in subsequent releases, with no possibility of being restored. If you need ISO 14443-A-4 emulation, try installing Flipper Zero firmware version 0.89.0 and using the CLI.

https://github.com/flipperdevices/flipperzero-firmware/issues/3276

bettse, 2025-02-22:
> with no possibility of being restored. 

that's a bit presumptuous. I believe they only used them for testing, so they didn't add them back after the NFC API refactor an year-plus ago.

 > If you need ISO 14443-A-4 emulation, try installing Flipper Zero firmware version 0.89.0 and using the CLI.

That doesn't really help me with designing an app for the current firmware.  Plus, using the CLI isn't programmatic emulation at the code level.

I'm more than happy to wait for them to have time to do what needs to be done for support.