flipperdevices/flipperzero-firmware #4060: ISO 14443-4 NXP - Mifare DESFire EV2 4k can only be read but NOT RIGHT emulated.

Killer74-hub, 2025-01-03:
### Describe the bug.

This NFC Tag can only be read, all Details will be displayed but when I emulate, nothing Happens. The Reader Just blinks Green and doesnt beep. Which indicates IT detects something, but Not the right NFC Signal.

Tag Typ: ISO 14443-4 NXP - Mifare DESFire EV2 4k

Technologies available: IsoDep, NfcA, NdefFormatable

Serialnumber: 04:10:5D:02:F5:72:80

ATQA: 0x0344

SAK: 0x20 

ATS: 0x80

Tag creator: SALTO/ KS

Green Tag.

Flipper Zero Emulating logs will come soon. 

### Reproduction

1. Read the NFC Tag
2. Save and Emulate or Emulate directly
3. Done

### Target

NFC(what should I Put Here?)

### Logs

```Text
Will come soon
```

Killer74-hub, 2025-01-03:
[image]
[image]
[image]
[image]
[image]

Killer74-hub, 2025-01-03:
Logs:

Killer74-hub, 2025-01-03:
[image]

bettse, 2025-01-08:
Your description said "Save and Emulate or Emulate directly", but you chose "Emulate UID", correct?

bettse, 2025-01-16:
> i also tried the other one.

Which one is the other one?  Are you are saying you saw a plain "Emulate" option?

Killer74-hub, 2025-01-16:
> > i also tried the other one.
> 
> Which one is "the other one"?  Are you are saying you saw a plain "Emulate" option?  
> 

Yes. Correct.

skotopes, 2025-01-16:
@Killer74-hub I think you misunderstand something, because current DESFire support only includes partial reading + saving for  UID emulation.

DESFire is quite decent in terms of technology it uses, so we are not planning to add emulation support at this point