flipperdevices/flipperzero-firmware #4061: mismatch between button menu and button panel

Akiva-Cohen, 2025-01-05:
### Describe the enhancement you're suggesting.

Currently button menu passes the event type to callbacks when a button is pressed, while button panel only calls the callback on a short press. In my opinion button pannel should be changed to be consistent with button menu.

skotopes, 2025-01-16:
Yep, sounds reasonable

Akiva-Cohen, 2025-01-16:
> Yep, sounds reasonable

I made a pr to solve the issue, but I'm not sure how to link it to the issue

skotopes, 2025-01-16:
Yep I saw it. It will be included in next release

Akiva-Cohen, 2025-01-16:
> Yep I saw it. It will be included in next release

this is my first time contributing to something like this. does release mean like some block of prs being added to dev or like is it just going to stay as a pr until main updates?

Akiva-Cohen, 2025-01-24:
Sorry to bug you I'm just wondering about that.

skotopes, 2025-01-27:
@Akiva-Cohen It's waiting for next merge cycle

Akiva-Cohen, 2025-02-21:
my original fix had bugs but you just merged my new one, so this is now fixed.

closed 2025-02-21 by Akiva-Cohen