flipperdevices/flipperzero-firmware #4067: infrared remote crashes the device when frequency is set to 76000

Didgitalpunk, 2025-01-14:
### Describe the bug.

infrared remote crashes the device when frequency is set to 76000, regardless of the signal output path chosen (flipper, A7)

[line 16 of furi hal interface](https://github.com/flipperdevices/flipperzero-firmware/blob/7291e6bd46c564400cdd3e9e7434f2c6e3a5ff28/targets/furi_hal_include/furi_hal_infrared.h#L16) specifies a max frequency of 56KHz.

Using a modified firmware with just that value set to 76000 instead of 56000 makes the flipper no longer crash, but still freezes before any command is sent when a button is pressed in the IR app.

### Reproduction

example code to run:

> Filetype: IR signals file
> Version: 1
> #
> #
> # LEGO Spybotics Remote Control (4232rc)
> #
> #
> #
> # CH 1
> #
> #
> name: Ch1_OrangeNop_YellowNop
> type: raw
> frequency: 76000
> duty_cycle: 0.330000
> data: 1040 208 208 208 208 624 208 416 208 208 1040 416
> #

expected behaviour: IR remote app sends the signal when button is pressed then stops when the button is depressed

actual behaviour: flipper crashes with a furi_check failed message on bootup when button is pressed,
                               OR
                           if using a modified version of furi_hal_infrared.h with line 16 set to 76000, flipper freezes and needs to be restarted

### Target

Infrared

Didgitalpunk, 2025-01-14:
After a bit of talk on the discord infrared channel (see [from here](https://discord.com/channels/740930220399525928/954422774141710366/1328614681362956298) ), this appears to be a hardware issue cause by how the signal is generated, specifically by using 2 hardware timers and Direct Memory Access (see furi_hal_infrared_configure_tim_pwm_tx and everything surounding INFRARED_DMA_TIMER)

skotopes, 2025-01-16:
System limitation comes from how infrared uses timers and DMA channels: long story short 1MHz carrier will generate 1M DMA transaction, which is close to what Flipper MCU designed to handle.

Interesting thing: previous max value was chosen based on infrared app performance when it was in MCU flash memory. After we moved apps to SD this limit is no longer actual.

I've bumped limit to ~1MHz, but there is no guarantee that it will always behave as expected(it may fail under specific type of load).

Didgitalpunk, 2025-01-20:
Awesome! Thanks for taking care of this so quickly!

Very interesting to know that moving things to the SD card pushed back the limitations a but!, makes you wonder what else could get a boost in performance out of that change!

PR #4070 (2025-02-12): Infrared: increase max carrier limit
# What's new

- Infrared: increase max carrier limit to 1MHz
- Fix #4067

# Verification 

- Create remote with carrier 1000000

# Checklist (For Reviewer)

- [ ] PR has description of feature/bug or link to Confluence/Jira task
- [ ] Description contains actions to verify feature/bugfix
- [ ] I've built this code, uploaded it to the device and verified feature/bugfix
closed 2025-02-12 by skotopes