flipperdevices/flipperzero-firmware #4072: Bad USB feature ignores keyboard layout setting (DE-CH)

ThWunderlin, 2025-01-18:
### Describe the bug.

When using the Bad USB feature with DE-CH keyboard layout selected, the Flipper Zero appears to still send US keyboard scancodes instead of using the selected layout.

### Reproduction

1. Set Flipper Zero Bad USB keyboard layout to DE-CH
2. Connect to MacBook Pro with Swiss German keyboard layout (DE-CH)
3. Run the default Bad USB demo script or a simple test script
4. Observe character mapping issues

Expected behavior:
- Flipper Zero should send correct scancodes matching the selected DE-CH layout
- Special characters should be correctly mapped for Swiss German layout

Actual behavior:
- Flipper Zero sends US keyboard scancodes regardless of selected layout
- This causes incorrect character mapping on Swiss German MacOS:
  - ! becomes +
  - " becomes à
  - § becomes ç
  - ( becomes -
  - ) remains )
  - = becomes ^
  - ? becomes _
  - \ becomes 
  - @ becomes ä
  - # becomes *

Test Environment:
- Device: MacBook Pro
- OS: MacOS with Swiss German (DE-CH) keyboard layout
- Flipper Zero Firmware: [Your firmware version]
- Also tested with Unleashed firmware with same results

Additional Notes:
- Same behavior observed with both official and Unleashed firmware
- Changing Flipper Zero keyboard layout setting has no effect on the output

### Logs

```Text

```

doomwastaken, 2025-03-10:
Greetings, we were able to reproduce the issue, will be added to backlog

ThWunderlin, 2025-03-10:
Thank you for looking into this! I'm glad to hear that you were able to reproduce the issue. I really appreciate it being added to the backlog. Looking forward to the fix!

rseufert, 2026-10-16:
Measured this on a Flipper Zero running 1.5.1 against a Mac, and the picture is a bit different from "the layout setting has no effect":

- With `de-CH.kl` selected, the Flipper does send the Swiss German key positions: Shift+¨ for `!`, Shift+2 for `"`, Shift+8/9/0 for `( ) =`, Shift+' for `?`, Shift+1 for `+`, and z/y swapped. I captured the raw key codes on the Mac with a system-wide event tap (those don't depend on the Mac's input source) and ran them through the macOS Swiss German keymap with `UCKeyTranslate`: `! " ( ) = ? + - _ z y` all come out correctly.
- What does go wrong on a Mac are the AltGr characters. `de-CH.kl` follows the Windows Swiss layout, where `@` is AltGr+2 and `\` is AltGr+`<`; on macOS those combinations give `“` and `ﬁ`. `[ ] { } ~` are on other Option keys too, and macOS swaps the ISO `<` key with `^`. Nine ASCII characters in total.

That is the same Windows-vs-Mac difference the repo already handles for German, French and Italian with the `*-mac.kl` variants, and Swiss had no Mac variant. #4471 adds `de-CH-mac.kl` and `fr-CH-mac.kl`, generated from the macOS keylayout data; with `de-CH-mac.kl` selected, a test string covering all 33 affected and unaffected punctuation characters decodes correctly under the Mac Swiss German keymap on the same setup.

@ThWunderlin if you still see `!` turning into `+` on current firmware, that would be a separate problem (it means the US table was in use), and it would help to know the firmware version and whether the layout shows as de-CH on the BadUSB screen while the script runs.