flipperdevices/flipperzero-firmware #4073: Crash with RCP app

gaellalire, 2025-01-18:
### Describe the bug.

When an RCP app is launched and not exited, the flipper crash with a fury message and pairing is lost.
You should close the app gracefully when connection is lost.
I tested with Sub-Ghz app but I guess it is for all RPC app.

Bluetooth connection is not reliable, you should manage lost connections properly.

### Reproduction

1) Launch an app in RPC mode in bluetooth (probably same issue with USB) with a phone
2) Disconnect (put the flipper in microwave for example)
3) Check that flipper crashed with an exception 

### Logs

```Text

```

doomwastaken, 2025-01-21:
Greetings, @gaellalire. Just to be sure are you launching app via our companion or via own app in RPC mode with BLE transport?

gaellalire, 2025-01-21:
Hello I created an App which call Sub-Ghz in RPC mode.
https://play.google.com/store/apps/details?id=fr.gaellalire.rf_pelilo

I just tested it in USB too, got same error: Flipper crashed and was rebooted (furi_check failed). But at least in USB you don't lose the pairing.

doomwastaken, 2025-03-10:
I was able to reproduce issues with incorrect use of the rpc (crash is still wrong), it is currently in backlog