flipperdevices/flipperzero-firmware #4081: Flipperzero crashes when GPS is attached to GPIO

dariusan, 2025-01-25:
### Describe the bug.

It looks like since the last firmware flipper zero crashes when a GPS is plugged to the GPIO connectors and no application reads the NMEA data which is send to UART. One has to reset the flipper all the time.

### Reproduction

1. plug a GPS board
2. don't run the GPS app
3. wait for Flipper to crash

### Logs

```Text

```

hedger, 2025-02-03:
@dariusan  Try disabling expansion protocol detection on that UART: settings > expansion modules > listen UART. Set to NULL. Check if the issue could still be reproduced.

dariusan, 2025-02-03:
@hedger This setting actually works. When I set it to "None" it does not crash anymore. What is the drawback if this setting?

hedger, 2025-02-04:
It is responsible for plug-and-play compatibility of external modules by implementing a special protocol. Anyway, looks like a bug.

aaronjamt, 2025-07-01:
I was able to trigger this bug once. It doesn't appear to be a total crash, since I unplugged the power cable and the RGB LED turned off, even after the crash. However, I do see a bunch of the following error in the logs, maybe it's just a matter of following its directions and increasing the stack size?
`225383 [E][ExpansionSrvWorker] Stack watermark is too low 204 < THREAD_STACK_WATERMARK_MIN. Increase stack size.`