flipperdevices/flipperzero-firmware #4083: device does not charge to 100%

gazirov, 2025-01-26:
### Describe the bug.

Short summary:
• The or it shuts down when the battery level is above 5%.
• I tried fully discharging the battery multiple times, but that didn’t help.
• I reset the Gauge, then recharged and discharged again, but the problem remains.
FW 1.1.2

I tested the battery capacity using an Imax B6 charger, and it turned out to be around 800 mAh (while the nominal capacity is 2100 mAh). The battery was swollen and pushing against the device casing. You can also see the remaining capacity in the device info under the parameter “Gauge Capacity Remain = 829.”

The battery health calculation is not working.
The battery percentage is calculated based on the nominal capacity (2100 mAh). The device counts how many milliamps went in and out rather than measuring the actual voltage on the battery.
If it charges from 0% to 41% but then reports that charging is complete at 4.1 V, that indicates the battery capacity has significantly dropped and should probably be replaced.

[image]

[image]

### Reproduction

 tried fully discharging the battery multiple times

### Target

 battery level

### Logs

```Text

```

skotopes, 2025-02-05:
Sadly health feature is not very reliable. Health numbers reset when battery fully discharged(battery goes into protection mode). We may change gauge configuration to store this data in its eeprom in future.

Also please contact support if you still have warranty.

closed 2025-02-05 by skotopes

gazirov, 2025-02-05:
The problem is not with calculating the battery's health, but with the charge level. Despite the fact that the battery's voltage is 4.1 volts, which corresponds to 100%, the flipper shows that the battery is at 49%, which is incorrect.

skotopes, 2025-02-05:
It is.
Just the way it work probably is not obvious: gauge calculates capacity, capacity then used to report charge percent, and then this percent reported to us. We just show what gauge reports.

gazirov, 2025-02-06:
It is not correct to use the gauge calculates capacity for determining the battery's charge level, because as batteries degrade their capacity decreases, and this will result in the flipper either not charging to 100% or shutting down while there is still 20% charge left. The gauge calculates capacity should be used specifically for assessing the battery's health, not for displaying the current charge level.

Imagine that over time your phone stopped charging to 100% and instead only reached 80%, or conversely, it charged to 100% but then suddenly shut off at 20%, with these values constantly fluctuating. For the user, that would be unpredictable behavior—a poor user experience. 
If I were to replace the battery now with a non-original one that, for example, has a lower capacity, I would again encounter the problem of an incorrect calculation of the battery’s charge level, since the system is hard-coded for 2100 mAh, and the charge level is calculated based on that value.

skotopes, 2025-02-06:
@gazirov properly configured gauge will do its job correctly.

Replacing battery with non original is not intended way of using flipper gauge, so please don't complain about that.

Also you can customize firmware with correct gauge config.