flipperdevices/flipperzero-firmware #4086: Region Mhz locks are incorrect for country code IM

omalmighty, 2025-01-29:
### Describe the bug.

Country code IM is in Region 1 which should have access to 433Mhz but is locking me out. 

### Reproduction

Be in country code IM
run a frequency at 433.92Mhz
get lock out message

### Logs

```Text

```

hedger, 2025-01-31:
Please provide links to government-issued regulatory documents proving that current limits are wrong.

omalmighty, 2025-01-31:
Link to regulatory website highlighting the Isle of Man extends the uk regulations:
https://www.cura.im/spectrum/#:~:text=The%20UK%20Wireless%20Telegraphy%20Act,of%20Man%20under%20that%20legislation.

gsovereignty, 2025-09-06:
Good to know that the creators of the Flipper are bootlicking Statists.