flipperdevices/flipperzero-firmware #4089: Bug in Manual Read: effects slix nfc, possibly others

FOSSBOSS, 2025-01-31:
### Describe the bug.

When you select a manual Read NFC if the slix tag is not found, the read process never times out.

### Reproduction

1.switch on.
2. navigate to NFC
3. read
4.Manual read
5. select slix nfc 
6. press read.
7. wait for the moon cycle to complete when your battery runs out. or re-flash your firmware. you can not escape the sequence, and it never times out. 

### Logs

```Text

```

hedger, 2025-02-03:
@doomwastaken please confirm if that can be reproduced

doomwastaken, 2025-02-25:
Greetings, we were not able to do this on official firmware, could you specifiy what firmware did you use and whether issue persists on 1.2 release