flipperdevices/flipperzero-firmware #4107: Custom Parity/Stop Bits for Serial

luken11, 2025-02-13:
### Describe the enhancement you're suggesting.

From what I can gather, serial connectivity is set at 8-N-1:
https://github.com/flipperdevices/flipperzero-firmware/blob/429c0dd387e82fa43a9c094e158ca3a3bb8d7aff/targets/f7/furi_hal/furi_hal_serial.c#L277-L279

I was looking at communicating with a device that expects 8-E-2, and wondered if it would be possible to expose the stop and parity settings in a similar way as the baud rate is, to allow for communications there? Unless I have missed something else that would achieve this goal

portasynthinca3, 2025-02-13:
Hi! We've already had this in our internal task tracker for some time. It's in the works.

portasynthinca3, 2025-02-20:
Implemented in #4121.

closed 2025-02-20 by portasynthinca3