flipperdevices/flipperzero-firmware #4134: FeliCa Poller Enumerate Through All Service Codes

zinongli, 2025-03-02:
### Describe the enhancement you're suggesting.

All of the following is under the assumption that we only care about the first system of a FeliCa tag. And we are reading without encryption.

Currently the NFC app's FeliCa poller only reads one service code. Seen [here](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/lib/nfc/protocols/felica/felica_poller.c#L207-L208).

But a lot of FeliCa card supports other and more than one service code. And it's only two byte: 65535 possibilities. It wouldn't be too crazy time-wise to enumerate through all of them in one read. A for loop to enumerate service code, within which a while loop to read until either SF reaches non-zero, would suffice. 

But this would require some major changes in how a FeliCa's content is formatted. As it will have dynamic size (0 to 65536 services theoretically readable, within each having 0-256 of blocks of data). The current way DESFire is handled is a great example: for each app there are files. In paralell, for each service there are blocks. 

I would love to contribute and I have available tags to test on my own. But I would love to know if this idea is in line with the dev team's vision for FeliCa poller and what the ideal way of implementation would be. Also happy to talk on Discord for faster exchanges. I'm @Torron there. Cheers!

zinongli, 2025-03-02:
As an example, [here](https://github.com/luu176/Metroflip/blob/dev/scenes/plugins/suica.c#L331-L365)'s how I implemented reading two services (a for loop) and reading each service's blocks until SFs turn non-zero (a while loop).

My implementation there only goes through two sevice codes because I know them a priori, i.e. others have enumerated and rev eng'ed Suica card's service codes. But for native NFC app, as a general reading tool, enumerating through 65536 would be nice, as no one knows what's not discovered.

zinongli, 2025-03-02:
Update: I did some benchmark. Enumerating the entire service code took about 370 seconds. 

[Here](https://github.com/zinongli/Metroflip/blob/felica_enum_demo/scenes/plugins/suica.c#L321-L358)'s the code I used for testing. I used Metroflip to demo because I'm more familiar with the code base. But it's just a simple usage of `felica_poller_read_blocks`. And here's the log I get from executing this code on a Suica:

```
687394 [I][Metroflip:Scene:Suica] Tic: 687394
710359 [I][Metroflip:Scene:Suica] Serv 0FFF Toc: 710359
717363 [I][Dolphin] Flush stats
717364 [I][SavedStruct] Saving "/int/.dolphin.state"
717386 [I][DolphinState] State saved
733596 [I][Metroflip:Scene:Suica] Serv 1FFF Toc: 733596
756844 [I][Metroflip:Scene:Suica] Serv 2FFF Toc: 756844
780004 [I][Metroflip:Scene:Suica] Serv 3FFF Toc: 780004
803162 [I][Metroflip:Scene:Suica] Serv 4FFF Toc: 803162
826317 [I][Metroflip:Scene:Suica] Serv 5FFF Toc: 826317
849470 [I][Metroflip:Scene:Suica] Serv 6FFF Toc: 849470
872627 [I][Metroflip:Scene:Suica] Serv 7FFF Toc: 872627
895775 [I][Metroflip:Scene:Suica] Serv 8FFF Toc: 895775
918926 [I][Metroflip:Scene:Suica] Serv 9FFF Toc: 918926
942071 [I][Metroflip:Scene:Suica] Serv AFFF Toc: 942071
965215 [I][Metroflip:Scene:Suica] Serv BFFF Toc: 965215
988363 [I][Metroflip:Scene:Suica] Serv CFFF Toc: 988363
1011510 [I][Metroflip:Scene:Suica] Serv DFFF Toc: 1011510
1034658 [I][Metroflip:Scene:Suica] Serv EFFF Toc: 1034658
1057800 [I][Metroflip:Scene:Suica] Toc: 1057800
```

So the time it took to go through every service code (and dumping the blocks of data when available) took 1057800 - 687394 = 370406 ms, 370.406 seconds. 

Caveat: Most of the service codes, of course, didn't have any data and denied our reading request immediately. For five 5 service codes, there were available data and I had some light reading (< 20 blocks). So my benchmark isn't definite. The reading time could be longer or shorter. Still, Suica is quite rich in available data for reading without auth, so I would presume this 370-second time is already on the longer side of expected reading time.

baconwaifu, 2025-04-09:
There actually is a command to enumerate area and service codes with a much smaller search space: Command 0x0A "Search Service Code." The detailed documentation for it is presumably behind an NDA, (the public docs only have the name and number) but it wasn't terribly difficult to figure out how it works by sniffing what NXP's TagInfo (lite version? the one on play store) does compared to what it reports.

The only obvious method to enumerate the *size* of a service, though, is to simply "keep reading until the card throws an error" from what I can tell. Thankfully a multi-block read appears to return as much data as it can, even if you ask for more blocks than the service has.

TagInfo is able to enumerate and dump all accessible services from a pasmo/suica in ~5-10 seconds that way, *far* shorter than your experimental 6 minutes.

```
Raw service command (length and checksum omitted)
0A  ??  ??  ??  ??  ??  ??  ??  ??  00  00

byte 0: command number
bytes 1-8: IDm
byte 9: 'node index?' (Official term unknown; starts from zero, increment until error to scan all areas/services)
byte 10: ?? (could be the tail end of a little-endian u16?)

Raw response:
0B  ??  ??  ??  ??  ??  ??  ??  ??  00  08  BF  0F
byte 0: response code
bytes 1-8: IDm
bytes 9-10: little-endian "start of area"/"service code"
bytes 10-11: little endian "end of area" (only present for areas)
```

dogtopus, 2025-04-29:
> There actually is a command to enumerate area and service codes with a much smaller search space: Command 0x0A "Search Service Code." The detailed documentation for it is presumably behind an NDA, [...]

This. This is also what @nullableVoidPtr and I used [in the now abandoned FeliCa branch](https://github.com/dogtopus/flipperzero-firmware/blob/13d85dd7694609c470b585258a637f7aa923d86b/lib/nfc/protocols/felica.c#L897-L912) years ago.

baconwaifu, 2025-07-25:
The length byte should be 0x0C, not 0x0B. It appears to include the count byte itself in the length. Aside from that, the command itself is fine.

And while the search space is theoretically 0xFFFF at the protocol level, every existing service entry is sequential from zero, and it returns a service code of 0xFFFF (which is an invalid service code) if there are no further services, so you can stop the scan very quickly. (Also, every card has a much smaller limit for service count in practice due to storage constraints, although the exact sizes are probably under NDA)

zinongli, 2025-07-25:
> The length byte should be 0x0C, not 0x0B. It appears to include the count byte itself in the length. Aside from that, the command itself is fine.

[image]

Thank you!

zinongli, 2025-07-27:
Hi @dogtopus , I see that you have the emulation PR pending merge (for two months already?!). Do you still plan on reimplementhing this dump service code and area node feature in the future? If not, I might give it a try.

dogtopus, 2025-07-27:
@zinongli You can safely do that. Emulation and read are two separate things and I don't plan to also add FeliCa Standard service code enumeration to the reader.