flipperdevices/flipperzero-firmware #4143: RFID raw files - Securakey

john-rybka, 2025-03-09:
I just got a Flipper Zero and I was trying to copy the tag for common spaces of my building, but I fail constantly.

So the system used in my building is SecuraKey Radiokey, reader models are RKDT-SR-M, and the token type RKKTH-02.
So I am using the newest Momentum dev software (with SecuraKey feature) on my flipper, I'm able to read RFID from token, but emulating is not doing anything to a readers.

I got a RAW RFID from a token and I am hoping anyone can help me out with it

[Securakey raw.zip](https://github.com/user-attachments/files/19151456/Securakey.raw.zip)

doomwastaken, 2025-03-11:
Greetings, how does the key behave on official firmware?

closed 2025-03-11 by john-rybka

john-rybka, 2025-03-13:
> > Поздрави, как се държи ключът на официалния фърмуер?
> 
> HELLO, IT WORKS PERFECTLY

Sorry, but it’s still not working. 
I tried many times but with no success